ms-iot/vscode-ros #379: [bug] launch not work

DYFeng, 2021-01-27:
Linux: ubuntu 20.04
ROS 1: noetic
Python 3.8.5

plugin version:0.6.6

Version: 1.52.1
Commit: ea3859d4ba2f3e577a159bc91e3074c5d85c0523
Date: 2020-12-16T16:32:10.090Z
Electron: 9.3.5
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Linux x64 5.8.0-38-generic

# what is the bug
Can not run launch file

## Repro steps
Error message:

Error from roslaunch: Unable to load file XXXXX.launch: 'XmlLoader' object has no attribute 'ignore_unset_args'

seanyen, 2021-02-02:
@DYFeng  Thanks for reporting. Can you share your launch file? And it seemed to complain the launch file not with the expected attribute `ingore_unset_args`? Could it be launch file formatting issue?

hengjiUSTC, 2021-05-01:
same issue. My launch config
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Launch",
            "type": "ros",
            "request": "launch",
            "target": "/home/jiheng/Documents/simulation/src/simulation_launch_car/launch/navigation_for_two_laser.launch",
        },

        {
            "name": "(gdb) Launch",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/devel/lib/hello_vs_code/vs_talker",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": false,
            "MIMode": "gdb",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                }
            ]
        },
        {
            "name": "ROS: Attach",
            "request": "attach",
            "type": "ros"
        }
    ]
}
```

hengjiUSTC, 2021-05-01:
Run ROS: Launch in debug get:
```
Error from ros launch 
unable to load file "/home/jiheng/Documents/simulation/src/simulation_launch_car/launch/navigation_for_two_laser.launch
ROS PATH[0] = /opt/ros/melodic/share/ros
ROS PATH[1] = /opt/ros/melodic/share
```

lovelyyoshino, 2021-05-02:
Using vscode multi-file debugging found that roslaunch can not be found, which means that there is no corresponding path.So first, check to see if "c_cpp_properties.json" is included in the project's "includePath" :Such as "{$workspaceFolder} / SRC/robot_test/ * *", and then check whether to add the file path to the to / ~ /.bashrc and then source it. Finally, reopen VS Code and re-test.

hengjiUSTC, 2021-05-02:
Thanks it work after I add `source workspace_path/devel/setup.bash` to ~/.bashrc. But I feels it pretty wired to add work space path to my global bashrc, is there better way to let ROS launch know correct path?

lovelyyoshino, 2021-05-02:
Generally there is no good way, or you can try to map the files to the ROS folder. Docker can also be used to do related operations

ooeygui, 2021-06-08:
Does this bug still repro for you? It should be sourcing the environment correctly.

ooeygui, 2021-07-13:
Please reopen if you still reproduce this. My only guess is that the extension isn't detecting this as a valid workspace, therefore not sourcing the setup.bash.

closed 2021-07-13 by ooeygui

knorth55, 2022-01-04:
This issue may be solved by https://github.com/ros/ros_comm/pull/2217.

igricart, 2022-10-07:
I am having the same issue. But the error is related to a package that couldn't be found 
In the launch file, defined as
`<include file="$(find belt_loader_bringup)/launch/run.launch" pass_all_args="true"/>`

igricart, 2022-10-07:
[image]
I tried opening `vscode` from a sourced terminal (bash or zsh) and none of them worked for me.