ms-iot/vscode-ros #388: [bug] Unable to see variable values while debugging C++ node

Chulabhaya, 2021-02-15:
(Please add appropriate labels)
- Windows 10, Windows Subsystem for Linux (Ubuntu 20.04)
- Noetic

\<Version of the plugin\>
0.6.6

\<Copy the Version information from the `Help | About` menu\>
[image]

# what is the bug
I'm unable to see the values of variables in debug mode

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Try to debug the simple hello_world code from the VS Code tutorials: https://code.visualstudio.com/docs/cpp/config-linux
2. Instead of seeing what's contained in the vector, like shown in the link below, I see something like this: 
[image]

# expected behavior
What I would expect to happen is that I would see the values contained within the msg vector, as shown in this section of the tutorial: https://code.visualstudio.com/docs/cpp/config-linux#_step-through-the-code

# additional context

ooeygui, 2021-06-08:
This is occurring in the VSCode C++ debugger itself, not the ROS extension. However, I'm marking it as we might need to update configuration.

Hypothesis-Z, 2021-06-15:
I have the same problem. A temporary solution I found is to enter the command `-exec -enable-pretty-printing` in the DEBUG CONSOLE, but it does not work for a new launch. 

I'm expecting a new release to fix it.

ooeygui, 2021-06-15:
@Hypothesis-Z Awesome - Can you provide more details or a PR with the fix?

Hypothesis-Z, 2021-06-16:
@ooeygui I'm not a typescript programmer. Anyway, I guess appending a property `setupCommands` to the configuration may work. 
```json
"setupCommands": [
    {
        "description": "Enable pretty-printing for gdb",
        "text": "-enable-pretty-printing",
        "ignoreFailures": true
    }
]
```

https://github.com/ms-iot/vscode-ros/blob/d56efda20b054cb2f1dc6e0a19e89ebf9bd0553f/src/debugger/configuration/resolvers/ros1/launch.ts#L275-L284

ooeygui, 2021-06-16:
Awesome!!! Thank you for the pointer, I'll take a look in the next batch of vscode work.

closed 2021-07-13 by ooeygui