ms-iot/vscode-ros #399: [feature] Request for more user-friendly instruction

JaehyunShim, 2021-03-03:
**what is needed**

Hi, 

If you see this [link](https://microsoft.github.io/Win-RoS-Landing-Page/#), it says this extension supports ROS2, but I cannot find any instruction about it ([link](https://github.com/ms-iot/vscode-ros)). Something written there about ROS1 is also not really instruction but more like a list of functions for those who already know how to use it. Do you guys have any plan to provide more user-friendly instruction like [this instruction](https://ros-qtc-plugin.readthedocs.io/en/latest/_source/How-to-Install-Users.html)?

I use VSCode but I saw people use QTCreator for this reason when they work on ROS projects. I tried to do the same in VSCode googling but I couldn't even find proper materials about how to use it.  

Regards,
Jaehyun

JaehyunShim, 2021-03-04:
Hi @seanyen 

I am not sure if you are the right person to tag here, but I remember you gave lots of advice regarding MS things. Do you have any ideas or advice?

Regards,
Jaehyun

Bi0T1N, 2021-07-19:
If you just open VS Code and look at the core status it'll say its offline but if you run the `Start Core` function, the ROS2 system status changes to online. For that VS Code starts the [ROS2 Daemon](https://answers.ros.org/question/327348/what-is-ros2-daemon/).

To attach to a running ROS2 node you need to do the same steps as for [ROS1](https://github.com/ms-iot/vscode-ros/blob/master/doc/debug-support.md#attach):
1. Build your package(s) with `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Debug`
2. Open the folder of your ROS2 package in VS Code
3. Go to the `Run and Debug` window
4. Click on `create a launch,json file`
5. Select `ROS` and then `ROS: Attach`
6. The `launch.json` file with following content should be created:
  ```
  {
      // Use IntelliSense to learn about possible attributes.
      // Hover to view descriptions of existing attributes.
      // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
      "version": "0.2.0",
      "configurations": [
          {
              "name": "ROS: Attach",
              "request": "attach",
              "type": "ros"
          }
      ]
  }
  ```
7. Go back to the `Run and Debug` window, start debugging with `ROS: Attach`, select C++ or Python depending on your source code language, enter your node name and accept to run as super user. Now you can set break points in your source files and debug the node.

JaehyunShim, 2021-07-19:
This question was asked a long time ago and I somewhat figured out a solution. Thanks!

ooeygui, 2021-07-19:
I'll make the ROS2 instructions more explicit and add ROS2 specific walkthroughs.

ooeygui, 2022-10-13:
I've added videos here https://youtu.be/PBbEhRf8QjE

closed 2022-10-13 by ooeygui