ms-iot/vscode-ros #404: [bug] generates colcon "log" directory everytime opening vscode in new directories

shintarokkk, 2021-03-06:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  Ubuntu 20.04.2
- [ ] ROS 1: Dist
- [x] ROS 2: Foxy
- [x] colcon: 0.6.1-1 (version of python3-colcon-core)

\<Version of the plugin\>
0.6.7

\<Copy the Version information from the `Help | About` menu\>
?

# what is the bug
\<current behavior\>

When opening vscode with this extension enabled in new directories, "log" directory is created.

For example,
<details>
<summary>Behavior and generated log file</summary>
<pre>
shintarokkk:~ $ mkdir -p /tmp/aaa
shintarokkk:~ $ cd /tmp/aaa/
shintarokkk:/tmp/aaa $ code .
shintarokkk:/tmp/aaa $ ls
log
shintarokkk:/tmp/aaa $ ls -l log/
合計 4
-rw-rw-r-- 1 shintarokkk shintarokkk    0  3月  6 10:44 COLCON_IGNORE
lrwxrwxrwx 1 shintarokkk shintarokkk   11  3月  6 10:44 latest -> latest_list
lrwxrwxrwx 1 shintarokkk shintarokkk   24  3月  6 10:44 latest_list -> list_2021-03-06_10-44-38
drwxrwxr-x 2 shintarokkk shintarokkk 4096  3月  6 10:44 list_2021-03-06_10-44-38
shintarokkk:/tmp/aaa $ cat log/list_2021-03-06_10-44-38/logger_all.log 
[0.097s] DEBUG:colcon:Command line arguments: ['/usr/bin/colcon', 'list', '-p', '--base-paths', '/tmp/aaa']
[0.097s] DEBUG:colcon:Parsed command line arguments: Namespace(base_paths=['/tmp/aaa'], build_base='build', ignore_user_meta=False, log_base=None, log_level=None, main=<bound method ListVerb.main of <colcon_package_information.verb.list.ListVerb object at 0x7f61e4acfd90>>, metas=['./colcon.meta'], names_only=False, packages_above=None, packages_above_and_dependencies=None, packages_above_depth=None, packages_end=None, packages_ignore=None, packages_ignore_regex=None, packages_select=None, packages_select_build_failed=False, packages_select_by_dep=None, packages_select_regex=None, packages_select_test_failures=False, packages_skip=None, packages_skip_build_finished=False, packages_skip_by_dep=None, packages_skip_regex=None, packages_skip_test_passed=False, packages_skip_up_to=None, packages_start=None, packages_up_to=None, packages_up_to_regex=None, paths=None, paths_only=True, topological_graph=False, topological_graph_density=False, topological_graph_dot=False, topological_graph_dot_cluster=False, topological_graph_dot_include_skipped=False, topological_graph_legend=False, topological_order=False, verb_extension=<colcon_package_information.verb.list.ListVerb object at 0x7f61e4acfd90>, verb_name='list', verb_parser=<colcon_defaults.argument_parser.defaults.DefaultArgumentsDecorator object at 0x7f61e498e040>)
[0.119s] Level 1:colcon.colcon_core.package_discovery:discover_packages(colcon_meta) check parameters
[0.119s] Level 1:colcon.colcon_core.package_discovery:discover_packages(recursive) check parameters
[0.119s] Level 1:colcon.colcon_core.package_discovery:discover_packages(ignore) check parameters
[0.119s] Level 1:colcon.colcon_core.package_discovery:discover_packages(path) check parameters
[0.119s] Level 1:colcon.colcon_core.package_discovery:discover_packages(recursive) discover
[0.119s] INFO:colcon.colcon_core.package_discovery:Crawling recursively for packages in '/tmp/aaa'
[0.119s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extensions ['ignore', 'ignore_ament_install']
[0.119s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extension 'ignore'
[0.119s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extension 'ignore_ament_install'
[0.119s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extensions ['colcon_pkg']
[0.119s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extension 'colcon_pkg'
[0.119s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extensions ['colcon_meta']
[0.119s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extension 'colcon_meta'
[0.119s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extensions ['ros']
[0.119s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extension 'ros'
[0.134s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extensions ['cmake', 'python']
[0.134s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extension 'cmake'
[0.134s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extension 'python'
[0.134s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extensions ['python_setup_py']
[0.134s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa) by extension 'python_setup_py'
[0.134s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa/log) by extensions ['ignore', 'ignore_ament_install']
[0.134s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa/log) by extension 'ignore'
[0.134s] Level 1:colcon.colcon_core.package_identification:_identify(/tmp/aaa/log) ignored
[0.134s] Level 1:colcon.colcon_core.package_discovery:discover_packages(recursive) using defaults
</pre>
</details>

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. install ROS extension to VSCode
2. install colcon
3. make new directory 
4. open VSCode there

# expected behavior
\<what you would expect to happen\>
"log" directory of colcon would be generated.

# additional context
\<any additional information would be helpful for a better understanding\>

winstxnhdw, 2021-07-10:
Bump. I get this issue as well.

ooeygui, 2021-07-13:
Thank you for the report. This has been fixed and will be released shortly

closed 2021-07-13 by ooeygui