ms-iot/vscode-ros #414: Cannot launch RViz using remote extension

synaptic, 2021-03-22:
- [x] Windows: Windows 10
- [x] Linux:  Ubuntu 20.04
- [x] ROS 2: foxy

Plugin version: 1.54.3

# What is the bug
I'm not sure if this is a vscode-ros issue vs a vscode-remote-extensionpack issue, but I thought I would start here.  Launching Rviz on the Linux remote machine works in the Windows' VS Code's integrated terminal, assuming you have an X11 client running and you have exported the DISPLAY environment variable beforehand:
```
$ export DISPLAY=localhost:11.0
$ ros2 launch mypackage rviz.launch.py
$ # success!
```

rviz.launch.py:
```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    
    start_rviz_cmd = Node(
        package='rviz2',
        executable='rviz2',
        name='rviz2',
        output='screen')

    # Create the launch description and populate
    ld = LaunchDescription()

    # Add any conditioned actions
    ld.add_action(start_rviz_cmd)

    return ld
```

launch.json:
```
{
    "version": "0.2.0",
    "environment": [{"name": "DISPLAY", "value": "localhost:11.0"}],
    "configurations": [
        {
            "name": "Launch rviz",
            "request": "launch",
            "target": "/path/myrepo/install/mypackage/share/mypackage/launch/rviz.launch.py",
            "type": "ros",
        },   
    ]
}
```

When I add this launch file to launch.json, it does not work when invoked via the UI.  Instead the following error is thrown in the Windows' VS Code UI:
```
"Unable to open 'raise.c': Unable to read file'vscode-remote://ssh-remote+<long_string>/build/glibc-eX1tMB/glibc-2.31/sysdeps/unix/sysv/linux/raise.c'"
```
Example:
[image]

## Steps to reproduce
1. Have ROS2 installed on Linux machine
2. Open workspace on Windows machine using VS Code with `ms-iot.vscode-ros` and `ms-vscode-remote.vscode-remote-extensionpack` extensions installed on remote Linux machine.
1. Add rviz.launch.py file to workspace
2. Add launch config to launch.json
3. Run via "Run and Debug" > select "Launch rviz" from drop-down > "Start debugging" (green arrow)
4. Error

# Expected behavior
Rviz UI should come up on Windows machine like it does when launched from integrated terminal. 

# Additional context
I also set up my C:/Users/<username>/.ssh/config as described here: 
https://github.com/microsoft/vscode-remote-release/issues/4600

ooeygui, 2021-06-08:
Sorry for the delay in responding to this - I missed it when it first came in.

I believe this is actually a vscode issue; It looks like it is trying to debug a URL instead of a file; not sure what code path this is going through.

ooeygui, 2021-11-13:
Added debug launch filtering. I didn't want to exclude rviz or gazebo completely from debugging, as this enables debugging extensions and plugins; but this allows the user to remove them from consideration for debugging.
https://github.com/ms-iot/vscode-ros/pull/605

ooeygui, 2022-10-13:
Closing as we no longer attempt to attach by default with launch filtering.

closed 2022-10-13 by ooeygui