ms-iot/vscode-ros #428: [bug]

igricart, 2021-04-16:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Ubuntu/18.04)
- [x] ROS 1: Melodic
- [ ] ROS 2: Dist

\<Version of the plugin\>

\<Copy the Version information from the `Help | About` menu\>
Version: 1.55.2
Commit: 3c4e3df9e89829dce27b7b5c24508306b151f30d
Date: 2021-04-13T09:37:02.931Z
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Linux x64 5.4.0-70-generic
# what is the bug
\<current behavior\>
Neither Launch or Attach are working
## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Build with debug flags `catkin build --cmake-args -DCMAKE_BUILD_TYPE=Debug`
2. Define tasks in `launch.json`
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Attach",
            "type": "ros",
            "request": "attach"
        },
        {
            "name": "ROS: Launch",
            "type": "ros",
            "request": "launch",
            "target": "/home/imr/Documents/ps/ps_ws/src/belt_loader_behavior_tree/launch/dummy_launch.launch"
        }
    ]
}
```
3. Setup roslaunch file
```
<launch>
  <node pkg="belt_loader_behavior_tree" type="executor_tree" name="test_bt">
    <param name="tree_xml" value="/trees/dummy_tree.xml" />
  </node>
</launch>
```
4. Start debugger with `ROS:Launch`

# expected behavior
\<what you would expect to happen\>
Start node
# additional context
\<any additional information would be helpful for a better understanding\>
The roslaunch works just fine if I run manually:
```
roslaunch belt_loader_behavior_tree dummy_launch.launch
```

ooeygui, 2021-04-18:
roscore currently needs to be running for launch debugging to work.
https://github.com/ms-iot/vscode-ros/issues/156

However, there is an inprogress feature to automatically launch roscore https://github.com/ms-iot/vscode-ros/pull/429

igricart, 2021-04-19:
I have the same behavior even with roscore running

hengjiUSTC, 2021-05-01:
I got similar issue and detail in #379

lovelyyoshino, 2021-05-02:
Using vscode multi-file debugging found that roslaunch can not be found, which means that there is no corresponding path.So first, check to see if "c_cpp_properties.json" is included in the project's "includePath" :Such as "{$workspaceFolder} / SRC/robot_test/ * *", and then check whether to add the file path to the to / ~ /.bashrc and then source it. Finally, reopen VS Code and re-test.

ooeygui, 2021-07-13:
Fixed - we now build with symbols on Linux by default.

closed 2021-07-13 by ooeygui