ms-iot/vscode-ros #443: [bug] URDF preview no longer work after VS Code update

weizhuowang, 2021-05-06:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Dist/Version)
- [x] ROS 1: Dist
- [ ] ROS 2: Dist

\<Version of the plugin\>
v0.6.7

\<Copy the Version information from the `Help | About` menu\>
Version: 1.56.0
Commit: cfa2e218100323074ac1948c885448fdf4de2a7f
Date: 2021-05-04T21:58:14.757Z
Electron: 12.0.4
Chrome: 89.0.4389.114
Node.js: 14.16.0
V8: 8.9.255.24-electron.0
OS: Linux x64 4.15.0-142-generic

# what is the bug
\<current behavior\>
I was editing an URDF file and it was showing in the preview just fine. Then there were an automatic update of vscode. After that the the previewer stopped showing the model, only shows the ground plane and origin. Code was not modified. I tried rebooting the computer and restarting the vscode, non worked. I also tried several of the turtlebot3 urdf, seems like previewer only shows non-STL links.

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Open any urdf file in the previewer.
3. bug appears

# expected behavior
\<what you would expect to happen\>
Expecting all links to show up, including STL/DAE links.

# additional context
\<any additional information would be helpful for a better understanding\>

ooeygui, 2021-05-06:
Thank you for the report.

ooeygui, 2021-05-06:
Few Questions:
Are you launching vscode from a ROS environment?
Is the URDF file within a ROS node?
Has the ROS node been built?

weizhuowang, 2021-05-06:
- Are you launching vscode from a ROS environment?
No I launched vscode from start menu and opened the catkin folder manually
- Is the URDF file within a ROS node?
Yes, to be specific urdf file is located at: catkin_ws/src/project_name/package_name/urdf/xxxxx.urdf.xacro
- Has the ROS node been built?
Yes I invoked catkin_make and sourced devel/setup.bash beforee I preview the urdf. No luck.

ooeygui, 2021-05-07:
Looks like we are hitting some cross site scripting issue:
```
vscode-webview-resource:/https://749bab03-b507-4413-99b3-07e44713ba0e.vscode-webview-test.com/vscode-resource/file///xxxx/src/turtlebot3/turtlebot3_description/meshes/sensors/lds.stl net::ERR_FAILED
```

or 

```
Access to XMLHttpRequest at 'vscode-webview-resource:/https://749bab03-b507-4413-99b3-07e44713ba0e.vscode-webview-test.com/vscode-resource/file///c%3A/xxxxx/src/turtlebot3/turtlebot3_description/meshes/sensors/lds.stl' from origin 'vscode-webview://749bab03-b507-4413-99b3-07e44713ba0e' has been blocked by CORS policy: Cross origin requests are only supported for protocol schemes: http, data, vscode-file, chrome, chrome-untrusted, vscode-webview, https.
```

ooeygui, 2021-05-07:
This is going to require more investigation. It looks like the URI scheme that vscode used changed. but the docs haven't been updated. 

We used to be able to handle `vscode-resource:`. Now it looks like it is through https and `<guid>.vscode-webview-test.com`. However, replacement doesn't seem to find the local resources. 

Plot thickens.

ooeygui, 2021-05-07:
This seems to be related to this bug in vscode https://github.com/microsoft/vscode/issues/122951. I'll follow up internally.

ooeygui, 2021-05-07:
Still working on this.

ooeygui, 2021-05-17:
0.6.8 is in the pipeline, which includes a fix for this on latest vscode.

ooeygui, 2021-05-17:
Thank you for reporting, 0.6.8 is now live with the fix.

closed 2021-05-17 by ooeygui