ms-iot/vscode-ros #455: ROS Commands not working on VS remote

TaylorDale, 2021-05-21:
We have VSCode remote set up and running to develop on a remote Linux Ubuntu machine. After installing the ROS extension, all of the commands included do not work.

- ROS Melodic
- Windows 10 host, Ubuntu 18.04 client

EG: Command 'ROS: Update C++ Properties' resulted in an error (command 'ros.updateCppProperties' not found)

ooeygui, 2021-05-27:
Hi @TaylorDale 
Thank you for the report. Can you tell me what version of vscode and the ROS extension you are using? Also, what kind of client? Is VSCode opened on a ROS workspace? (Meaning that you've opened a ROS workspace folder, not just navigated to a folder which contains a ROS workspace).  Is the ROS extension installed in the remote container?

TaylorDale, 2021-05-28:
Hi, Thanks for the reply. I've just logged in today and it all of a sudden seems to be working again. Must have just needed a reboot or something. 

Much appreciated.

For posterity:

- VSCODE: 1.56.2, ROS EXT: 0.6.8
- AARM64, Nvidia Jetson AGX Xavier
- ROS workspace is directly opened
- Extension shows as installed ("Extension is enabled on "SSH:host")

closed 2021-05-28 by TaylorDale

ooeygui, 2021-05-28:
Thank you for following up - I do appreciate it.

Please do post back if you see any other issues. 

Also note, that if you are debugging C++, you'll need to manually build with symbols. I'm working on a fix to build automatically with symbols so that it works out of the box.