ms-iot/vscode-ros #463: [bug] When debugging, sometimes the launch file is opened

ooeygui, 2021-06-04:
# what is the bug
When debugging a ROS node with a lunch file, the launch.json file opens.

# Expect
It doesn't open... I just shut it anyway.

ooeygui, 2021-07-13:
This occurs with the vscode API:
https://github.com/ms-iot/vscode-ros/blob/7d43d124ab2fcde88d898890e8cff57a71fd30bb/src/debugger/configuration/resolvers/ros1/launch.ts#L302

I've reached out to the API owners to see if there's way to suppress this.

ooeygui, 2021-07-15:
This looks like the reason:
https://github.com/microsoft/vscode/blob/4d50d169012f4abfe867572c828e8a85002ad1e2/src/vs/workbench/contrib/debug/browser/debugService.ts#L448

ooeygui, 2021-11-13:
This looks like it is going to be a huge refactor of the debug system; I'm going to remove it from 0.7.0

ooeygui, 2021-12-01:
Happens always

andrejpan, 2024-03-21:
Quite an old issue, but it still persists nowdays with extension version 0.9.2

ooeygui, 2024-03-21:
@andrejpan It turns out that this is by design in vscode, I'm trying to find a workaround.
I suspect that if I create a debug group it will go away.

ooeygui, 2024-03-22:
@andrejpan Thanks for taking a peek!
The VSCode ROS Extension exposes it's own Debugger entry point, which 'redirects' to either C++ or Python.
This happens differently on ROS2 and ROS1: but looks like this:
https://github.com/ms-iot/vscode-ros/blob/814a8ef20ec8c04aa71fc6b01b021df78570ba35/src/debugger/configuration/resolvers/ros2/debug_launch.ts#L79

We currently pass 'undefined' to the first arg of the startDebugging API, which is expected if this is an unrooted debug.
https://code.visualstudio.com/api/references/vscode-api#debug

However (if I recall correctly), looking through the VSCode sources, it appears this has the unintended side effect of opening the launch file.

I suspect that passing a real root would prevent this from happening.