ms-iot/vscode-ros #465: How to allow multiple terminals on integrated vscode terminal ?

wbadry, 2021-06-06:
Hello,
I created a launcher file for `turtlesim_node` , `velocity_publisher` and `pose_subscriber`

```xml
<launch>
    <node pkg="turtlesim" type="turtlesim_node" name="turtlesim" respawn="true" />
    <node pkg="wbadry" type="pubvel_node" name="publish_velocity" required="true" output="screen" />
    <node pkg="wbadry" type="subvel_node" name="pose_subscriber" output="screen" launch-prefix="gnome-terminal --command" />
</launch>
```

My problem is that I had to create an external terminal using `launch-prefix="gnome-terminal --command"` to monitor both 'publisher' and 'subscriber` messages.

Is there any way to use multiple integrated terminals within `vscode` like running multiple `rosrun` nodes? In other words, can I use the `launch-prefix` attribute to instantiate the integrated terminal instance inside `vscode`?

ooeygui, 2021-06-07:
The VSCode ROS node extension exposes many commands through the VSCode command Pallet. You can access that using 'ctrl-shift-P'.  The command you want is 'ROS: Create Terminal'. Looking at the VSCode extension API, I don't see a way to enable this external launch back to the originating window. 

One thought, you can have a single launch file which includes your sub-launch files, and load them all at once.

wbadry, 2021-06-08:
> The VSCode ROS node extension exposes many commands through the VSCode command Pallet. You can access that using 'ctrl-shift-P'. The command you want is 'ROS: Create Terminal'. Looking at the VSCode extension API, I don't see a way to enable this external launch back to the originating window.
> 
> One thought, you can have a single launch file that includes your sub-launch files, and load them all at once.

Thanks, @ooeygui for your response. Unfortunately, this would be equivalent to calling `rosrun` multiple time 😄 

Thanks again for being dedicated to work on this 

[image]

ooeygui, 2021-06-08:
Hi @wbadry,
I think I see what you are trying to do. Unfortunately, vscode doesn't have this model. I am planning on adding rqt like functionality, which allows you to monitor specific topics; would that be useful?

wbadry, 2021-06-08:
@ooeygui it would be very indeed

ooeygui, 2023-05-04:
I changed title based on the chat inline