ms-iot/vscode-ros #472: [bug] Preview URDF Error loading webview

igor-markovic, 2021-06-15:
- Linux:  Ubuntu 20.04.2 LTS
- ROS 1: noetic
- ROS plugin version 0.6.8

- VSCode Help -> About:
Version: 1.57.0
Commit: b4c1bd0a9b03c749ea011b06c6d2676c8091a70c
Date: 2021-06-09T17:18:42.895Z
Electron: 12.0.9
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Linux x64 5.8.0-55-generic

# what is the bug
Error loading webview: Error: Could not register service workers: InvalidStateError: Failed to register a ServiceWorker: The document is in an invalid state..

## Repro steps
1. Open any urdf or xacro file in vscode
2. `ctrl` + `shift` + `p` ROS: Preview URDF
3. URDF Preview will be completely blank and error notification appears

May be related to #443, but they were using v1.56. Did something change in v1.57 of VSCode?

igor-markovic, 2021-06-15:
Issue no longer occurs after restarting computer.

closed 2021-06-15 by igor-markovic

ooeygui, 2021-06-15:
Thank you for reporting and following up. I'll keep my eye out for this - not sure where that error is coming from.