ms-iot/vscode-ros #496: [bug] ROS Extension detects every folder as a ROS workspace

stef264, 2021-07-12:
- [ ] Linux:  Ubuntu 20.04
- [ ] ROS 2: foxy

\<Version of the plugin\>
v0.6.8

Version: 1.58.0
Commit: 2d23c42a936db1c7b3b06f918cde29561cc47cd6
Date: 2021-07-08T06:53:55.113Z
Electron: 12.0.13
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Linux x64 5.10.0-1034-oem

# what is the bug
\<current behavior\>
Every folder i open with vscode gets detected as ROS workspace.

## Repro steps
1. Open a empty folder in vscode
2. ROS Extensions creates .vscode and log directories and files

It works with Windows 10 as intended, but on Ubuntu every folder gets detected as ROS workspace.

# expected behavior
no dirs and files created

# additional context
logger_all.log
~~~
[0.249s] DEBUG:colcon:Command line arguments: ['/usr/bin/colcon', 'list', '-p', '--base-paths', '/home/stef/test']
[0.250s] DEBUG:colcon:Parsed command line arguments: Namespace(base_paths=['/home/stef/test'], build_base='build', ignore_user_meta=False, log_base=None, log_level=None, main=<bound method ListVerb.main of <colcon_package_information.verb.list.ListVerb object at 0x7f3fe023df40>>, metas=['./colcon.meta'], names_only=False, packages_above=None, packages_above_and_dependencies=None, packages_above_depth=None, packages_end=None, packages_ignore=None, packages_ignore_regex=None, packages_select=None, packages_select_build_failed=False, packages_select_by_dep=None, packages_select_regex=None, packages_select_test_failures=False, packages_skip=None, packages_skip_build_finished=False, packages_skip_by_dep=None, packages_skip_regex=None, packages_skip_test_passed=False, packages_skip_up_to=None, packages_start=None, packages_up_to=None, packages_up_to_regex=None, paths=None, paths_only=True, topological_graph=False, topological_graph_density=False, topological_graph_dot=False, topological_graph_dot_cluster=False, topological_graph_dot_include_skipped=False, topological_graph_legend=False, topological_order=False, verb_extension=<colcon_package_information.verb.list.ListVerb object at 0x7f3fe023df40>, verb_name='list', verb_parser=<colcon_defaults.argument_parser.defaults.DefaultArgumentsDecorator object at 0x7f3fe00d3e80>)
[0.273s] Level 1:colcon.colcon_core.package_discovery:discover_packages(colcon_meta) check parameters
[0.273s] Level 1:colcon.colcon_core.package_discovery:discover_packages(recursive) check parameters
[0.273s] Level 1:colcon.colcon_core.package_discovery:discover_packages(ignore) check parameters
[0.273s] Level 1:colcon.colcon_core.package_discovery:discover_packages(path) check parameters
[0.273s] Level 1:colcon.colcon_core.package_discovery:discover_packages(recursive) discover
[0.273s] INFO:colcon.colcon_core.package_discovery:Crawling recursively for packages in '/home/stef/test'
[0.273s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extensions ['ignore', 'ignore_ament_install']
[0.273s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extension 'ignore'
[0.274s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extension 'ignore_ament_install'
[0.274s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extensions ['colcon_pkg']
[0.274s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extension 'colcon_pkg'
[0.274s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extensions ['colcon_meta']
[0.274s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extension 'colcon_meta'
[0.274s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extensions ['ros']
[0.274s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extension 'ros'
[0.284s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extensions ['cmake', 'python']
[0.284s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extension 'cmake'
[0.284s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extension 'python'
[0.284s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extensions ['python_setup_py']
[0.284s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test) by extension 'python_setup_py'
[0.284s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test/log) by extensions ['ignore', 'ignore_ament_install']
[0.284s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test/log) by extension 'ignore'
[0.284s] Level 1:colcon.colcon_core.package_identification:_identify(/home/stef/test/log) ignored
[0.284s] Level 1:colcon.colcon_core.package_discovery:discover_packages(recursive) using defaults
~~~

ooeygui, 2021-07-12:
Thank you for the remote - this is smilar to https://github.com/ms-iot/vscode-ros/issues/404/

ooeygui, 2021-07-13:
Thank you for the report. This has been fixed and will be released shortly.

closed 2021-07-13 by ooeygui