ms-iot/vscode-ros #513: [bug] launch.json for ros package appears to ignore args

CDietzel, 2021-07-27:
- Linux:  (Ubuntu 20.04.2 LTS)
- ROS 1: Noetic
- VSCode ROS Version: 0.6.8

Version: 1.58.2
Commit: c3f126316369cd610563c75b1b1725e0679adfb3
Date: 2021-07-14T22:30:16.440Z
Electron: 12.0.13
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Linux x64 5.8.0-55-generic

# what is the bug
I am trying to set up a launch configuration to run a ROS .launch file.  This launch file is designed to accept a few command line arguments (without which, the program will crash). The entire terminal command, when run correctly, should look like this: ```roslaunch /home/locobot/interbotix_ws/src/locobot_puppet_manager/launch/puppet_manager.launch robot_model:=locobot_wx250s use_base:=true use_camera:=true use_lidar:=true```

I have set up a launch.json file like this:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Launch",
            "request": "launch",
            "target": "/home/locobot/interbotix_ws/src/locobot_puppet_manager/launch/puppet_manager.launch",
            "type": "ros",
            "args": ["robot_model:=locobot_wx250s", "use_base:=true", "use_camera:=true", "use_lidar:=true"]
        }
    ]
}
```
However, when I press Ctrl+F5 to run the launch.json file, ROS is launched with no arguments. This results in ROS crashing with the same error message as if I had run ```roslaunch /home/locobot/interbotix_ws/src/locobot_puppet_manager/launch/puppet_manager.launch```

Ultimately, it seems like VSCode is ignoring the argument parameters specified in the launch.json file. 

## Repro steps
1. open a ROS package with a launch file that requires commandline arguments
2. set up a VSCode launch.json file with those arguments
3. see that the arguments are not passed to the launch file

# expected behavior
The arguments listed in launch.json should be passed to the launch file.

rickstaa, 2021-07-28:
Thanks for raising this issue. I also experience this problem in:

**Version:** 1.58.2
**Commit:** c3f126316369cd610563c75b1b1725e0679adfb3
**Date:** 2021-07-14T22:30:16.440Z
**Electron:** 12.0.13
**Chrome:** 89.0.4389.128
**Node.js:** 14.16.0
**V8:** 8.9.255.25-electron.0
**OS:** Linux x64 5.8.0-63-generic

and: 

**Version:** 1.59.0-insider
**Commit:** e437d424c5e6de97632f5513229c7ebc2cf27a32
**Date:** 2021-07-28T05:13:02.250Z
**Electron:** 13.1.7
**Chrome:** 91.0.4472.124
**Node.js:** 14.16.0
**V8:** 9.1.269.36-electron.0
**OS:** Linux x64 5.8.0-63-generic

I tried the following syntaxes:

```json
"args": ["control_type:=position_control"]
```

```json
"args": ["control_type",":=","position_control"]
```

```json
"args": ["control_type", "position_control"]
```

bxh1, 2021-08-20:
I meet the same bug

ooeygui, 2021-11-13:
We use 'arguments' internally - https://github.com/ms-iot/vscode-ros/blob/894a975708cad7c8687a0fc6d25a2c18d4672e85/src/debugger/configuration/resolvers/ros1/launch.ts#L26. 

I updated the doc in https://github.com/ms-iot/vscode-ros/pull/605.

closed 2021-12-27 by ooeygui