ms-iot/vscode-ros #537: vscode-ros: command 'ros.startCore' not found

coder-day2up, 2021-08-22:
hello, there is one issue when i use the vscode-ros plugin.
ubuntu version:14.04
vscode version:1.42
vscode-ros version:0.6.8
C/C++ plugin ,python plugin : installed for vscode-ros
(1)when i click right on the "src" directory to perform "caktin create package",there is no response. in fact, it should popup one bar for me to type the name of the newpackage.
(2) when i try to perform "ros.startcore" with ctrl+shift+p,it shows "command 'ros.startCore' not found" ,
above all,it seems the vscode -ros doesn't working.

Extension version: 0.6.2
VS Code version: Code 1.40.2 (f359dd69833dd8800b54d458f6d37ab7c78df520, 2019-11-25T14:54:40.719Z)
OS version: Linux x64 4.4.0-148-generic

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-9300H CPU @ 2.40GHz (2 x 2399)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: disabled_off<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: disabled_software<br>skia_renderer: disabled_off<br>surface_control: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: unavailable_off<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|1, 0, 0|
|Memory (System)|1.94GB (0.08GB free)|
|Process Argv|--no-sandbox --unity-launch|
|Screen Reader|no|
|VM|100%|
</details>

NovoG93, 2021-08-23:
Same for me for all commands provided by this extension.

Ubuntu 20.04
vscode version:1.59.1
vscode-ros version:0.6.8

chris-ferone, 2021-08-23:
I also have the same problem. None of the extension commands work. I get the  "command 'ros.startCore' not found"  error message.

v0.6.8 ROS exstension

**From VS code about menu:**
Version: 1.59.1
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-19T11:54:37.513Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Linux x64 5.11.0-27-generic

**From lsb_release -a :**
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal

maicon-cesar, 2022-10-09:
Unfortunately I'm getting this same error with a current version. None of the extension commands work. I get the "command 'ros.startCore' not found" error message.

v0.8.3 ROS extension

VS CODE
Version: 1.71.2
Commit: 74b1f979648cc44d385a2286793c226e611f59e7
Date: 2022-09-14T21:12:14.256Z
Electron: 19.0.12
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 5.15.0-48-generic
Sandboxed: No

Ubuntu 14.04.6 LTS  in a container  (system requirements)
Codename:       trusty

ooeygui, 2022-10-10:
@maicon-cesar I wanted to confirm your ubuntu install is 14.04? The VSCode extension supports Melodic forward.   I'd be happy to accept a PR, with community support for EOL distributions.

maicon-cesar, 2022-10-10:
Hello Lou,

Yes, Ubuntu 14.04 with ROS Indigo.
Nowhere have I read about some ROS1 version restriction to run with vscode extension.
It is a feature of our system to run in Ubuntu 14.04 and ROS Indigo e we would appreciate to use the VSCode as our development environment.

ooeygui, 2022-10-10:
@maicon-cesar I'd love to get more details about your system. 

The VS extension doesn't register most ROS handlers for the declared commands when it doesn't detect that it is a ROS workspace. This is why you can activate the command but get an error from VSCode when the command is executed. 

I've wanted to fix it so that the commands are always registered but gives more details about why the command can't be executed - like that it couldn't find a valid ROS install or catkin/colcon workspace. Fixing that path requires some surgery in a hot path.

However, how the VSCode extension looks for a distro is based on the ROS_DISTRO environment variable; once it determines it is a ROS install, it will they call catkin or colcon (and in some cases catkin tools).

However, at this time, I don't have resources to add features to ROS1, so asking for community support.

maicon-cesar, 2022-10-11:
It seems that the problem is to run inside a container, it works outside the container in a native Ubuntu 14.04 and with ROS Indigo.

ooeygui, 2022-10-11:
Thank you for the update. This comment makes me think it is a manifestation of https://github.com/ms-iot/vscode-ros/issues/753. It appears to be an issue with the login shell used to connect to the container.

ooeygui, 2023-05-04:
This has been corrected in 0.9.4

closed 2023-05-04 by ooeygui

maicon-cesar, 2023-08-22:
Thank you very much, I tested it and now it's working with pre-release 0.9.4.
Though it's only for Noetic on Ubuntu 20. It didn't work on an environment with Ubuntu 14 and Indigo. But we are migrating to using Noetic in the project.