ms-iot/vscode-ros #550: [bug] Python import could not be resolved for my own module

FSund, 2021-09-07:
(Please add appropriate labels)
- Linux:  Pop!OS 20.04
- ROS 1: Noetic
- ROS extension v0.6.9
- Python extension v2021.10.1200408502-dev

VSCode
Version: 1.59.1
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-19T11:54:37.513Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Linux x64 5.11.0-7620-generic

# what is the bug
Importing my local ROS package results in a warning in VSCode, even though the code runs without errors.

[image]

## Repro steps

    mkdir -p test_ws/src
    cd test_ws/src
    source /opt/ros/noetic/setup.bash
    catkin_init_workspace
    git clone https://github.com/FSund/ros_vscode_issue_example
    cd ..
    catkin build
    source devel/setup.bash
    code .

- Open either `src/example/test/test.py` or `src/other_package/src/executable.py`

# expected behavior
I would not expect the import to give any warnings, since the code works fine.

# additional context

`.vscode/settings.json`

    {
        "python.autoComplete.extraPaths": [
            "/home/filip/ros/test_ws/devel/lib/python3/dist-packages",
            "/opt/ros/noetic/lib/python3/dist-packages"
        ]
    }

cmuell89, 2022-04-15:
Do you think the issue could be related to how the dist-packages are symlinked?

ooeygui, 2022-04-15:
@cmuell89 I'm not quite sure why this isn't working. I need to set up a repro and get some help from my friends on the python extension team - the ROS extension is likely not configuring the project correctly, but not quite sure what's missing.

FSund, 2022-04-19:
This issue is still present on my machine with the latest Python/ROS extension. Running `ROS: Update Python path` does not resolve it.

- ROS extension v0.8.2
- Python extension v2022.5.11051003
- Pylance extension v2022.4.2-pre.1

```
# settings.json
{
    "python.autoComplete.extraPaths": [
        "/home/filip/ros/test_ms_iot_ws/devel/lib/python3/dist-packages",
        "/opt/ros/noetic/lib/python3/dist-packages"
    ],
    "python.analysis.extraPaths": [
        "/home/filip/ros/test_ms_iot_ws/devel/lib/python3/dist-packages",
        "/opt/ros/noetic/lib/python3/dist-packages"
    ]
}
```

cmuell89, 2022-07-21:
@ooeygui 

Just curious if you've been able to look into this issue at all? It definitely serves as a productivity hindrance when working heavily with my own ROS packages. Appreciate your time!

ooeygui, 2022-07-21:
@cmuell89 I'm sorry, I have not been able to look at this. I don't think I'll have bandwidth in the next few weeks to investigate VSCode issues.
I'd love some help if someone has some expertise in Python.

mbcel, 2022-10-03:
Same Problem here. Any updates on this?

ooeygui, 2022-10-03:
No updates to share just yet - I have a large pending release of VSCode (0.9.0), then will be looking at issues like this.

JohnnyMudcrab, 2023-02-21:
Don't really know the mechanics behind it, but if I also put the include path in "python.analysis.extraPaths" it works for me. Here is an example for your case:

`.vscode/settings.json`

```
{
    "python.autoComplete.extraPaths": [
        "/home/filip/ros/test_ws/devel/lib/python3/dist-packages",
        "/opt/ros/noetic/lib/python3/dist-packages"
    ],
    "python.analysis.extraPaths": [
        "/home/filip/ros/test_ws/devel/lib/python3/dist-packages",
        "/opt/ros/noetic/lib/python3/dist-packages"
}
```

Some of the packages are already included there in my case, but apparently not all.

yurirocha15, 2023-03-15:
I am also experiencing this issue. I added the paths to `.vscode/settings.json` as shown below:
```json
{
    "python.autoComplete.extraPaths": [
        "/home/yuri/catkin_ws/devel/lib/python3/dist-packages",
        "/opt/ros/noetic/lib/python3/dist-packages",
    ],
    "python.analysis.extraPaths": [
        "/home/yuri/catkin_ws/devel/lib/python3/dist-packages",
        "/opt/ros/noetic/lib/python3/dist-packages",
    ],
}
```
However, VScode still can't find my custom package imports (it works for custom messages). 
Also, if I run `pyright` in my ROS project manually, it reports the same error (`reportMissingImports`). As PyLance uses `pyright` internally, this might be related to the issue.

argenos, 2023-07-12:
Also running into this, same configuration as the last two comments above mine, and also matches `PYTHONPATH` in my terminal. No idea if this provides any hints, but there seems to be a difference between opening the entire workspace and just the package containing the module. In the latter case, pylance is able to resolve the imports.