ms-iot/vscode-ros #552: [bug] ROS2 status not visible in remote docker container

Serafadam, 2021-09-09:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Ubuntu/18.04, Jetson Xavier NX)
- [ ] ROS 1: Dist
- [x] ROS 2: Galactic

\<Version of the plugin\>
Released on
6/28/2019, 23:09:22
Last updated
8/25/2021, 22:18:38
Identifier
ms-iot.vscode-ros

VSCode About info:
Version: 1.60.0
Commit: e7d7e9a9348e6a8cc8c03f877d39cb72e5dfb1ff
Date: 2021-09-01T10:43:02.136Z
Electron: 13.1.8
Chrome: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.11.0-34-generic snap
# what is the bug
I want to debug some nodes inside docker container on a remote host. I set up my launch configuration correctly (the way that I debug the program on my local pc), but after running the debug, I receive a timeout message:
[image].

 I cannot start the ROS2 system by command (when I run the command to start it, nothing happens).
If I run the program in terminal, it runs just fine, but ROS2 status badge still doesn't indicate that ROS is running (it is on VSCode that is opened on my local PC). If I open the terminal, and run `ros2 topic list`, the daemon starts (though it is not visible from my local PC). Launch files don't start the daemon (or at least it is not detected).

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Have a container with ROS built from source on a remote PC.
2. Connect to that PC via SSH
3. Run the container manually in shell
4. Connect to SSH with VSCode
5. Attach VSCode to running container via Docker extension
6. Run some nodes in terminal (talker/subscriber for example)
7. See if the ROS2 status indicates that the system is running

# expected behavior
When connecting to the remote container, I'd like to be able to debug the nodes without manually starting ros2 daemon

# additional context
Can this be because ROS2 was built from source?

grishalipenko, 2021-11-20:
Hi!
I didn't build ROS2 from source. And I had same problem with container running on my PC:
1. Run container in linux terminal.
2. Attach through Remote-Containers extension in VSCode to running container.
[image]
[image]
3. Attach to this container through linux terminal.
4. Make `$ ros2 topic list` in linux terminal and have block:
[image]
5. Make `Close Remote Connection` in VScode and got error for `$ ros2 topic list` in linux terminal:
```
$ ros2 topic list
Traceback (most recent call last):
  File "/opt/ros/foxy/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.9.9', 'console_scripts', 'ros2')()
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2topic/verb/list.py", line 38, in main
    with NodeStrategy(args) as node:
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/node/strategy.py", line 52, in __enter__
    self._daemon_node.__enter__()
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/node/daemon.py", line 116, in __enter__
    methods = self._proxy.system.listMethods()
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1109, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1450, in __request
    response = self.__transport.request(
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1153, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1165, in single_request
    http_conn = self.send_request(host, handler, request_body, verbose)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1278, in send_request
    self.send_content(connection, request_body)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1308, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python3.8/http/client.py", line 1247, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1007, in _send_output
    self.send(msg)
  File "/usr/lib/python3.8/http/client.py", line 947, in send
    self.connect()
  File "/usr/lib/python3.8/http/client.py", line 918, in connect
    self.sock = self._create_connection(
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
```

Ok.
1. Rerun container in linux terminal.
2. Attach to this container through linux terminal.
3. Check topic list in linux terminal:
```
$ ros2 topic list
/parameter_events
/rosout
```
4. Check ros daemon status:
```
$ ros2 daemon status
The daemon is running
```
5. Attach through Remote-Containers extension in VSCode to running container.
6. Recheck topic list in linux terminal:
```
$ ros2 topic list
/parameter_events
/rosout
```
And have no block and errors now!

ooeygui, 2023-05-04:
This has been corrected - it was a race condition on startup.

closed 2023-05-04 by ooeygui