ms-iot/vscode-ros #554: [feature] Implement ROS: Clean

ooeygui, 2021-09-09:
**what is needed**
Implement `ROS: Clean` which cleans a ROS Workspace.

**why is it needed**
When switching between distros or upgrading build components, a previously built workspace can be out of date. Cleaning the workspace today is a manual process. Having a quick way to build clean would be very useful.

ooeygui, 2022-01-07:
Would love for this to be pulled into colcon https://github.com/colcon/colcon-core/issues/37 for cleaning up a workspace.

ruffsl, 2022-12-15:
The colcon clean extension has now been released:

https://github.com/colcon/colcon-core/issues/37#issuecomment-1353287688 

See the extension's repo readme for more documentation:

https://github.com/colcon/colcon-clean