ms-iot/vscode-ros #557: HOW TO DEBUG  NODE WITH ROS2

GaHooooo, 2021-09-14:
I try the way  which supported by ROS1
https://github.com/ms-iot/vscode-ros/blob/master/doc/debug-support.md#attach
but it can't work when i have a try in ROS2
dose any document  with HOW TO DEBUG  NODE WITH ROS2
hope anyone can reply me ,THANKS

ooeygui, 2021-09-14:
Hi there,
I'm working on videos and will post them on the walkthrough page. In the mean time:
VSCode ROS works on launch files - you'll need to create a launch file which launches just the nodes you want to debug. In the debug tab, create a configuration - you'll see `ROS` as one of the items. chose that, then launch, and select your new launch file.

You can then debug normally.

polyhobbyist, 2022-04-26:
I've posted videos on using the VSCode ROS extension, including with ROS2 - https://youtu.be/PBbEhRf8QjE

ooeygui, 2022-10-13:
Closing with video links.

closed 2022-10-13 by ooeygui