ms-iot/vscode-ros #567: Debugging ros2 launch file

mnissov, 2021-09-30:
I see a lot of content regarding how one can setup vscode for debugging a ros node. Is it possible to debug the launch file itself?

This may not have been so interesting in ros1, but in ros2 with the added complexity of the python launch file format I feel it could be useful if it doesn't yet exist. In short, is it currently possible to debug the launch file itself in ros2 foxy and if so how?

Using: ros2 foxy and Ubuntu 20.04 LTS.

ooeygui, 2021-11-13:
I'll look into it - I don't see why this can't be supported.

MTDzi, 2022-02-11:
Hey @ooeygui, any updates on this?

ooeygui, 2022-02-11:
@MTDzi Thanks for checking - it is a super useful feature, but I have not started looking at it. I'd be happy to collaborate if someone wants to take a crack at this.

CM134, 2023-03-24:
Hey, I am just checking in. Are there now ways to debug launch files in ros2? (using humble)

ooeygui, 2023-03-24:
Thanks for the interest in this :-)

I'm working on it in https://github.com/ms-iot/vscode-ros/tree/feature/debug_launch_files.

I might split this into two work items - one which is about debugging the launch file itself, without launching the ROS nodes; and the second which launches them as well.

ooeygui, 2023-05-04:
We now support debugging the launch file itself - although it does not actually launch ROS nodes at the moment.

closed 2023-05-04 by ooeygui

martinakos, 2023-08-30:
I'm trying to debug a launch_test file, which is similar to a launch file. Would it work using this feature?

ooeygui, 2023-08-30:
> I'm trying to debug a launch_test file, which is similar to a launch file. Would it work using this feature?

Hi there,
Can you post a link to a sample of your launch file?
This currently works with python based launch files.

martinakos, 2023-08-31:
The launch_test file I use is of this type
https://github.com/ros2/launch_ros/blob/rolling/launch_testing_ros/test/examples/talker_listener_launch_test.py
This is to run integration tests, launching nodes like in a launch.py file but you can perform some tests on the outputs of the nodes. 

The test can be run like this 
`launch_test  launch_testing_ros/test/examples/talker_listener_launch_test.py`

it would be great if I could put breakpoints either in the tests or in the nodes code and step through with vscode.

ooeygui, 2023-08-31:
@martinakos Thanks for the clarification. Yes, you can debug the tests themselves. I have a video on setting up launch debugging. https://www.youtube.com/watch?v=A6ABRdL0ckg&feature=youtu.be.

ROS Test is set up similarly, although I don't have a video on it.

(This feature is about walking through launch files without actually launching ROS Nodes)

I'll create an issue for documenting test debugging.