ms-iot/vscode-ros #588: [bug] Unable to start debugging. Launch options string provided by the project system is invalid. Unable to determine path to debugger.  Please specify the "MIDebuggerPath" option.

GarethZhang, 2021-11-01:
(Please add appropriate labels)
- [ ] Linux:  20.04
- [ ] ROS 1: Noetic

Plugin version 0.6.9

# Bug description
I am developing ROS nodes in a remote Docker container.

I followed the document at https://github.com/ms-iot/vscode-ros/blob/master/doc/debug-support.md to set up my debugging environment to debug via a ROS launch file.

In the Run and Debug window of VSCode, when I run `ROS: Launch` from the upper left corner of VSCode window, the process runs but terminates with the following message:
`Unable to start debugging. Launch options string provided by the project system is invalid. Unable to determine path to debugger.  Please specify the "MIDebuggerPath" option.`

This is my `launch.json`:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Launch",
            "request": "launch",
            "target": LAUNCH_FILE_PATH,
            "type": "ros",
            "preLaunchTask": "make_debug", // <<< This is the task that will run before debugging starts
        }
    ]
}
```

This is my `tasks.json`:
```
{
    "version": "2.0.0",
    "tasks": [
        {
            "label": "make_debug",
            "type": "catkin_make",
            "args": [
                "--directory",
                "${workspaceFolder}",
                "-DCMAKE_BUILD_TYPE=Debug", // This extra argument enables built with debug symbols
            ],
            "problemMatcher": [
                "$catkin-gcc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            },
        },
    ]
}
```

I'm new to VSCode, it would be great if I can get more perspectives or a potential solution out of this. Thank you!

## Repro steps
Please see the above bug description

# expected behavior
Allow me to set breakpoint and debug

MentosSeetoo, 2021-11-14:
It seems that `vscode-ros` is running based on GDB debugger.
If you are using WSL2 in windows, you may need to check whether the GDB debugger is installed or not.
Or you can simply try to install GDB debugger by:
`sudo apt install gdb`

ooeygui, 2021-11-15:
I should add some code to detect this when creating a launch file.

GarethZhang, 2021-11-15:
@MentosSeetoo Thanks! Got it working by installing GDB debugger!

closed 2021-11-15 by GarethZhang