ms-iot/vscode-ros #606: [bug] ROS attach "your 131072x1 screen size is bogus, expect trouble"

sink0918, 2021-11-13:
- [ ] Windows: (Version)
- [x] Linux:  (Ubuntu/20.04 WSL2)
- [ ] ROS 1: Dist
- [x] ROS 2: Galactic

\<Version of the plugin\>
v0.6.9

\<Copy the Version information from the `Help | About` menu\>
[image]

# what is the bug
Using ROS attach to debug with an error msg:
[image]

## Repro steps
Simply start `ROS: Attach`

ooeygui, 2021-11-13:
This was fixed in https://github.com/ms-iot/vscode-ros/pull/544. It will be release with 0.7.0, which I'm working toward by 11/19/21.

closed 2021-11-13 by ooeygui

chenjun00, 2021-12-15:
This problem happens on lastest VSCode again:
[image]

ooeygui, 2021-12-15:
0.7.0 has not been released yet; I closed it in anticipation. I'm working on one final validation pass before pushing it.

Sorry for the delay, this had some big changes!

zhiwu111, 2024-09-13:
I'm having the same issue now, how do I fix it?
我现在也遇到了同样的问题，我应该怎么解决？