ms-iot/vscode-ros #611: [cannot attach remote machine for debug]

Existever, 2021-11-17:
(Please add appropriate labels)
- [ ] Linux:  18.04
- [ ] ROS 2: Eloquent Elusor | November 22nd, 2019

\<Version of the plugin\>
v0.6.9

\<Copy the Version information from the `Help | About` menu\>

# what is the bug
hi,i can't attach to remote machine to debug ros2 node 

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1.  set the configuration in  .vscode/launch.json as below
```
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Attach",
            "type": "ros",
            "request": "attach"
        },
]
```
2. run ros2  `ros2 launch demo demo_launch.py`
3. press F5 ,select runtime type `python`-->selecect process `demo_launch`

# expected behavior
expected to  attach to demo_launch and debug ros2 node,but i get nothing but jump to   .vscode/launch.json

polyhobbyist, 2021-11-30:
In order to debug remotely, you'll need to use the SSH remote extension to remotely connect. Once VSCode is connected, you'll need to install the ROS extension remotely. You can then debug from that context.

ooeygui, 2021-12-27:
Closing - this requires the SSH extension and installing the ROS extension remotely

closed 2021-12-27 by ooeygui