ms-iot/vscode-ros #623: [bug] URDF Preview does not handle relative paths.

ooeygui, 2021-12-10:
When a URDF includes a relative path to an object, the URDF fails to render.

gonultasbu, 2021-12-10:
referring to #261 
one example relative path to this would be:
`./meshes/mesh_cylinder.obj`

The same failure also occurs with absolute system paths such as:
`/mnt/c/Users/burak/Documents/GitHub/diffnav/data/urdf/simple_ackermann/meshes/mesh_cylinder.obj` 
or 
`file:///mnt/c/Users/burak/Documents/GitHub/diffnav/data/urdf/simple_ackermann/meshes/mesh_cylinder.obj`

MingFengHill, 2021-12-14:
Error details are not available when loading UFDF files. Is there a way to get the error log？

gonultasbu, 2021-12-14:
@MingFengHill please refer to this comment https://github.com/ms-iot/vscode-ros/issues/261#issuecomment-990101301