ms-iot/vscode-ros #624: [bug] debugger with ROS: Launch failed with NodeParamsException

SubaruArai, 2021-12-10:
(Please add appropriate labels)
- [x] Linux:  (Ubuntu 16.04)
- [x] ROS 1: Kinetic

\<Version of the plugin\>
v0.6.9

\<Copy the Version information from the `Help | About` menu\>
> Version: 1.63.0 (user setup)
> Commit: 7db1a2b88f7557e0a43fec75b6ba7e50b3e9f77e
> Date: 2021-12-07T06:26:56.179Z
> Electron: 13.5.2
> Chromium: 91.0.4472.164
> Node.js: 14.16.0
> V8: 9.1.269.39-electron.0
> OS: Windows_NT x64 10.0.18362

# what is the bug
\<current behavior\>
when starting debugger with launch, it fails with the following message in a popup window:
```
Command failed: roslaunch --args /package_name /home/user_name/catkin_ws/src/package_name/launch/file_name.launch
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 237, in main
    node_args.print_node_args(options.node_args, args)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/node_args.py", line 108, in print_node_args
    args = get_node_args(node_name, roslaunch_files)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/node_args.py", line 189, in get_node_args
    args = create_local_process_args(node, machine)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/node_args.py", line 254, in create_local_process_args
    raise NodeParamsException(str(e))
  NodeParamsException: package_name
  ROS path[0] = /opt/ros/kinetic/share/ros
  ROS path[0] = /opt/ros/kinetic/share
```
(Please substitute `package_name`, `user_name`, `file_name` for their actual names)

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. write launch.json according to the guide
```json
{
    "configurations": [
        {
            "name": "ROS: Launch",
            "type": "ros",
            "request": "launch",
            "target": "/home/user_name/catkin_ws/src/package_name/launch/file_name.launch"
        }
    ]
}

``` 
2. press `start debugging`(green play button) in `run and debug`
3. the error appears

# expected behavior
\<what you would expect to happen\>
It just works

# additional context
\<any additional information would be helpful for a better understanding\>
- I first tried this feature yesterday and it just worked. Today it won't no matter which launch file I choose.
- The launch file itself is fine, and using the same command in a terminal will correctly output the commands from `roslaunch --args`.
- I tried to roll back the version of vscode to 1.62.3 or this extension's version to 0.6.8, but none solved the problem.
- I am running ROS inside a virtual machine with virtualbox and connecting through ssh.

robotrancher, 2021-12-24:
Kinetic is not supported and EOL. Can you try on a currently supported ROS distro? 
Check out - https://dlu.github.io/ros_clock/index.html

SubaruArai, 2021-12-24:
> Kinetic is not supported and EOL. Can you try on a currently supported ROS distro? Check out - https://dlu.github.io/ros_clock/index.html

Ok, I'll try but it will take some time.
I'm closing this issue as of now.

closed 2021-12-24 by SubaruArai