ms-iot/vscode-ros #632: [bug] Debug launch stuck with LifeCycleNode (Ros2, foxy)

quentingllmt, 2021-12-20:
(Please add appropriate labels)
- [ ] Linux:  (Ubuntu/20.04)
- [ ] ROS 2: Foxy

\<Version of the plugin\>
0.6.9

\<Copy the Version information from the `Help | About` menu\>
```
Version: 1.60.2 
Electron: 13.1.8
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
```
# what is the bug
\<current behavior\>
When starting a debug section from a launch file, process is stuck before debug starts. (Only the green arrow and red square are active, progress bar is loading indefinitely).

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
Try to launch a ROS2 launch.py with a LifecycleNode in it.

# expected behavior
\<what you would expect to happen\>
Debug starts !

jandaa, 2022-01-07:
Same issue here

ooeygui, 2022-01-21:
Looking into this. I'm using https://github.com/ros2/demos/tree/foxy/lifecycle as a baseline

ooeygui, 2022-01-21:
Initial investigation shows that our "Launch file dumper" is getting stuck parsing the LifecycleNode in the launch file.

closed 2022-01-22 by ooeygui