ms-iot/vscode-ros #634: Can't detect mymsg.h generate from mymsg.msg[bug]

zhixin-zhao, 2021-12-23:
- [ ] Linux:  ubuntu18.04
- [ ] ROS 1: melodic

v0.6.9

After I make my custom message by mymsg.msg and make proper change in CMakeLists.txt and package.xml, I import the messge in my node c++ file by adding `#include "mypkg/mymsg.h"`. However vscode editor show errors on the new added import line.
I ignore the error and catkin_make the project , everything works fine. So I believe it is a bug from the extension.

zhixin-zhao, 2021-12-23:
Modifying `includePath` in the c_cpp_properties.json file fixes the problem. But I don't know why vscode can't do it automatically (I do have source the devel/setup.bash before open code editor in termial)

robotrancher, 2021-12-24:
It can - it's just code. Feel free to submit a pull request.
The VSCode ROS extension does have a command to update the C++ (and python) properties. 0.7.0 includes full browse database for ROS

ooeygui, 2021-12-27:
The 0.7.0 release improves intellisense, including adding the local include paths. However, you have to manually update it.

I'm going to change this to a feature request for automatically updating an existing C++ property

zhixin-zhao, 2021-12-28:
> The 0.7.0 release improves intellisense, including adding the local include paths. However, you have to manually update it.
> 
> I'm going to change this to a feature request for automatically updating an existing C++ property

I should have pulled a request. Appreciate it!

zhixin-zhao, 2021-12-28:
> It can - it's just code. Feel free to submit a pull request. The VSCode ROS extension does have a command to update the C++ (and python) properties. 0.7.0 includes full browse database for ROS

Yeah, now I realise it is better to pull a request not a bug report. Thanks