ms-iot/vscode-ros #639: [bug] When trying to Launch Debugging using launch.json for ROS, There is no trace of ROS

Coderx7, 2021-12-28:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [×] Linux:  (Dist/Version) Ubuntu 20.04
- [×] ROS 1: Dist
- [ ] ROS 2: Dist

ROS extension version : ```0.7.0```
VSCode Version: 

> 1.63.2
> Commit: 899d46d82c4c95423fb7e10e68eba52050e30ba3
> Date: 2021-12-15T09:39:46.686Z
> Electron: 13.5.2
> Chromium: 91.0.4472.164
> Node.js: 14.16.0
> V8: 9.1.269.39-electron.0
> OS: Linux x64 5.11.0-34-generic snap

# what is the bug
Based on the explanation provided in [Automatic creation of a launch.json with ROS Launch support ](https://github.com/ms-iot/vscode-ros#automatic-creation-of-a-launchjson-with-ros-launch-support), after selecting the link to create a `launch.json` file, in the 
drop-down menu, there should be a ROS option, but nothing shows up!. even if I go ahead and create a C++ `launch.json`
there is no trace of ROS related entries there.   
(all other functionalities seem to work though!) here is how it looks : 
[image]

## Repro steps
how to reproduce the bug. Please attach sample files or github repo if appropriate.
1.create a simple node, something as simple as 
```cpp
#include <ros/ros.h>
#include <std_msgs/String.h>
#include <iostream>
#include <sstream>

void callback(std_msgs::String message)
{
    ROS_INFO("Recieved : [%s]\n", message.data.c_str());
}

int main(int argc, char** argv)
{
    ros::init(argc, argv, "our_first_basic_node_listener");
    ros::NodeHandle node_handle;
    
    // ros::Subscriber subscriber = node_handle.subscribe("our_topic", 1000, callback);
    auto callback_fn = [](std_msgs::String::ConstPtr message)
    {
        ROS_INFO("[Listener] Recieved : [%s]\n", message->data.c_str());
    };
    ros::Subscriber subscriber = node_handle.subscribe<std_msgs::String>("our_topic", 1000, callback_fn);
    ros::spin();
    return 0;
}

```
2. Follow the steps provided in [Automatic creation of a launch.json with ROS Launch support ](https://github.com/ms-iot/vscode-ros#automatic-creation-of-a-launchjson-with-ros-launch-support)
3. at step 4 you should see the issue as no ROS related option shows up  

# expected behavior
The ROS option should show up

# additional context

ooeygui, 2022-01-22:
While debugging https://github.com/ms-iot/vscode-ros/pull/656, I figured out that VSCode is filtering the options when not registered with a debug event. ROS now shows up.

Thank you so much for the detailed report! I really appreciate the effort put into creating it. I loved the animated gif.

ooeygui, 2022-01-22:
I pinned this as an example of an incredibly helpful bug!

closed 2022-01-22 by ooeygui

JohnnyMudcrab, 2022-04-28:
Hi,

i am still experiencing the exact same issue:

[×] Linux: (Dist/Version) Ubuntu 20.04
[×] ROS 2: Dist (foxy)

ROS extension version : `0.8.2`
VSCode Version:
> Version: 1.66.2
> Commit: dfd34e8260c270da74b5c2d86d61aee4b6d56977
> Date: 2022-04-11T07:49:24.808Z
> Electron: 17.2.0
> Chromium: 98.0.4758.109
> Node.js: 16.13.0
> V8: 9.8.177.11-electron.0
> OS: Linux x64 5.14.0-1033-oem snap