ms-iot/vscode-ros #648: [bug] Some args of tasks `colcon test` is invalid.

chenjunnn, 2022-01-15:
- [ ] Windows: (Version)
- [x] Linux:  Ubuntu 20.04
- [ ] ROS 1: Dist
- [x] ROS 2: Galactic

\<Version of the plugin\>
0.7.0

# what is the bug
Some args of tasks `colcon test` is invalid.
```json
{
	"type": "colcon",
	"args": [
		"test",
		"--symlink-install",
		"--event-handlers",
		"console_cohesion+",
		"--base-paths",
		"/home/chenjun/ros2_ws",
		"--cmake-args",
		"-DCMAKE_BUILD_TYPE=RelWithDebInfo"
	],
	"problemMatcher": [
		"$catkin-gcc"
	],
	"group": "test",
	"label": "colcon: test"
}
```
I think `"--symlink-install",` `"--cmake-args",
		"-DCMAKE_BUILD_TYPE=RelWithDebInfo"` should be removed.

ooeygui, 2022-01-22:
Thank you for the suggestion. However:
* RelWithDebInfo is required for debugging.
* Symlink install prevents colcon from copying during build, which improves dev iteration.

Do either of these cause problems for your use case?

closed 2022-01-22 by ooeygui

chenjunnn, 2022-01-22:
> Some args of tasks `colcon test` is invalid.

@ooeygui  What I said is `colcon test`, not `colcon build`. Please try using these arguments with colcon test, and you will get:
```bash
> Executing task: colcon test --symlink-install --event-handlers console_cohesion+ --base-paths /home/chenjun/ros2_ws --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo <

usage: colcon [-h] [--log-base LOG_BASE] [--log-level LOG_LEVEL] {build,extension-points,extensions,graph,info,list,metadata,test,test-result,version-check} ...
colcon: error: unrecognized arguments: --symlink-install
终端进程“zsh '-c', 'colcon test --symlink-install --event-handlers console_cohesion+ --base-paths /home/chenjun/ros2_ws --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo'”已终止，退出代码: 2。
```