ms-iot/vscode-ros #676: [bug] Rosmaster URI export does not work

ZygaOGC, 2022-02-21:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Ubuntu/20.04)
- [x] ROS 1: Dist
- [ ] ROS 2: Dist

\<0.8.0\>

# what is the bug
Can not launch debugging with a remote rosmaster.

## Repro steps
1. Export ROS_MASTER_URI variable in the pre-launch task.
2. Launch debugging 

# expected behavior
Node under debugging should connect to the rosmaster available at exported URI

robotrancher, 2022-02-21:
Unfortunately, VSCode itself does not propagate environment variables between tasks, so this is lost well before the ROS extension activates. Instead you should launch vscode from a pre-initialized environment, or ensure that you are using ros from a root install. Check out this video for more details: https://www.youtube.com/watch?v=N2vqBvPQdhE

ooeygui, 2022-02-21:
I'm adding support for specifying a setup script by full path instead of just attempting to inferring. This resolves some issues where ROS isn't installed in a standard location.

ooeygui, 2022-02-27:
Support for arbitrary script loading has been added in https://github.com/ms-iot/vscode-ros/pull/678

closed 2022-02-27 by ooeygui