ms-iot/vscode-ros #685: [bug] Can't launch debugger without specifying an "args" entry in launch.json

bearrista, 2022-02-26:
WSL2: Ubuntu 18.04
ROS 1: Melodic

Release 0.8.1

Version: 1.65.0-insider (user setup)
Commit: eeeab304b9ac777b4a8dddf55f66ffebd9e4e7f2
Date: 2022-02-22T14:29:15.294Z
Electron: 16.0.9
Chromium: 96.0.4664.174
Node.js: 16.9.1
V8: 9.6.180.23-electron.0
OS: Windows_NT x64 10.0.19044

# what is the bug
I believe in my recent patch to resolve #645 I inadvertently added a new bug. I now can't launch my test node WITHOUT an "args" entry in the launch.json

[image]

## Repro steps
1. Attempt to debug a ROS1 node whose launch file doesn't require args and ensure there is no "args" entry in the launch.json

# expected behavior
Debugger should launch

# additional context
I'm a complete noob at TypeScript and JavaScript, sorry. Also I clearly didn't perform my due diligence on that patch.

ooeygui, 2022-02-27:
Investigating

ooeygui, 2022-02-27:
I see what happened. `args` is not the correct variable - `arguments` is. This is declared on the LaunchRequest object - however, it isn't passed in correctly to the roslaunch.
I'm still testing.

ooeygui, 2022-02-28:
Fix released in 0.8.2

closed 2022-02-28 by ooeygui