ms-iot/vscode-ros #696: [bug]

joseantoniobecerrapermuy, 2022-03-24:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [X] Linux:  (Ubuntu 20.04)
- [X] ROS 1: Noetic
- [ ] ROS 2: Dist

\<Version of the plugin\>
0.8.2

\<Copy the Version information from the `Help | About` menu\>
Versión: 1.65.2
Commit: c722ca6c7eed3d7987c0d5c3df5c45f6b15e77d1
Fecha: 2022-03-10T14:33:52.251Z
Electrón: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux arm64 5.13.0-30-generic

# what is the bug
\<current behavior\>
devel/setup.bash takes precedence over install/setup.bash when using an install space even if ros.rosSetupScript is used. This causes that packages belonging to the ROS workspace are not found (source devel/setup.bash when the workspace was created to use an install space does not properly modify ROS environment variables).

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Configure a workspace using "catkin config --init --install"
2. Open the workspace in Visual Studio Code
3. Modify ros.rosSetupScript to point to workspace/install/setup.bash
4. Workspace's packages are not found when trying to run / debug a launch file.

# expected behavior
\<what you would expect to happen\>
Packages should appear in the same way they do when not using an install space.

# additional context
\<any additional information would be helpful for a better understanding\>
1. Packages in /opt/ros are always correctly found.
2. If the workspace is created using "catkin config --init", it works.
3. If I set ros.rosSetupScript does not work.
4. If I set ros.rosSetupScript AND I delete devel/setup.bash file, then it works!

polyhobbyist, 2022-04-26:
Are you are using catkin tools?

joseantoniobecerrapermuy, 2022-04-27:
> Are you are using catkin tools?

Yes, it's implicit in the commands I have written in my comments...

ooeygui, 2023-05-04:
This has been corrected

closed 2023-05-04 by ooeygui