ms-iot/vscode-ros #710: [feature] Support Debugging ROS2 Launch Files

polyhobbyist, 2022-04-26:
**what is needed**
Launch files in ROS2 are Python scripts which execute parameterized ROS2 nodes. Additionally, ROS2 lifecycle nodes are maintained by the launch solution.

**why is it needed**
As these grow in complexity, they need to be debugged.

adamsj-ros, 2022-04-26:
ROS2 python launch files are probably the most difficult source files in ROS2 to debug. First, there’s a lot going on when you started them including other nodes and associated source. Secondly, there’s almost no we’ll-understood way to debug them including prints, pdb, and gdb. +1

chilisam, 2022-10-26:
Any updates on this?

ooeygui, 2022-10-26:
@chilisam This is a popular request; it will be prioritized soon. However, I don't have an exact date on the release.

polyhobbyist, 2023-03-24:
[image]

ooeygui, 2023-05-04:
This has been implemented

closed 2023-05-04 by ooeygui