ms-iot/vscode-ros #738: [bug] ROS2 status always showing Offline

ubuntuslave, 2022-06-10:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Dist/Version)
  - Ubuntu 20.04
- [ ] ROS 1: Dist
- [x] ROS 2: Dist
  - foxy

- Version of the plugin
2/26/2022, 23:12:23

# what is the bug
Even when the ROS2 daemon is running, the status is "Offline".
[image]

## Alternative bug
Also, I'm not able to launch anything. It times out waiting for the ROS daemon to start:
[image]

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Installed ROS2 foxy from binaries
2. Installed the `vscode-ros` plugging (Last updated: `2/26/2022, 23:12:23`)
3. Start the ROS2 daemon from the Cmd Pallete in VS Code ("ROS: Start")
4. Run the "ROS: Show Status" and expect to see the status page pop up as "Offline" (as shown in the screenshot)
5. However, the `rus2 daemon status` from the Terminal shows it's "running" (as shown in the screenshot)

# expected behavior
I would expect to see the proper status and list of topics in the page

# additional context
\<any additional information would be helpful for a better understanding\>

ubuntuslave, 2022-06-10:
**Update**: If I build the extension from source (from this forked repo:  https://github.com/ms-iot/vscode-ros), it works as expected

ooeygui, 2022-06-12:
Thank you confirming that the current sources are working, that helps narrow down the issue. I do need to push a new release which I'll work on this week.

ooeygui, 2022-06-30:
This issue has come up a couple of times recently; but I haven't been able to reproduce it.

Lakmusluk, 2022-09-01:
I had the exact same behavior.
My problem was that I set the `ROS_DOMAIN_ID` to a different value in my .bashrc file. 

Maybe this helps to dig deeper into the issue.

ooeygui, 2022-09-01:
@Lakmusluk Thank you for the suggestion - I do need to improve support for Domain ID. This currently attempts to capture the environment domain ID. I should have this as a property in the environment
https://github.com/ms-iot/vscode-ros/blob/0a5d534027fd0ec8cc631a52ef96a94a9d6f4ea3/src/ros/ros2/ros2-monitor.ts#L13

Tenfi, 2022-10-16:
same issue. 
try `ros:stop` then `ros:start`, it works for me.

ooeygui, 2023-05-04:
This has been corrected

closed 2023-05-04 by ooeygui