ms-iot/vscode-ros #745: [vscode extension 0.8.2 ubuntu foxy ROS2 start command is not ok]

yeyanlei, 2022-06-30:
(Please add appropriate labels)
- [ ubuntu 20.04] Linux:  (Dist/Version)
- [foxy ] ROS 2: Dist

\<Version of the plugin\>
vscode extension 0.8.2
\<Copy the Version information from the `Help | About` menu\>

# what is the bug
\<current behavior\>
i have a test, foxy is not ok, humble is ok using ros2 start command. vscode extension v0.8.2,   I have posted codes, screenshots and videos.
## Repro steps
\<how to reproduce the bug. Please atta
[image]
ch sample files or github repo if app

https://user-images.githubusercontent.com/36154470/176689134-cd874830-0947-4ead-b244-c821fa04dec2.mp4

ropriate.\>
[ros2_vscode.zip](https://github.com/ms-iot/vscode-ros/files/9020156/ros2_vscode.zip)

[image]

# expected behavior
\<what you would expect to happen\>

# additional context
\<any additional information would be helpful for a better understanding\>

ooeygui, 2022-06-30:
Thank you for posting the videos. One piece of context I'm missing is where is ROS2 installed?
VSCode ROS extension needs to source the ROS environment by itself, so it tries to source `/opt/ros/<distro>/setup.bash`. Is ROS in that location? 

If you have installed ROS elsewhere, you'll need to configure the startup script using the ros.rosSetupScript option in the global or workspace settings, pointing to the specific start script. 

(I'm not able to reproduce this, on my ubuntu 20.04/Foxy setup, which is why I'm asking for more details)

Thanks!

yeyanlei, 2022-07-04:
i have installed ros, and i have added env    (source /opt/ros/foxy/setup.bash in ~/.bashrc ).  Publishing node is running normally.
[ros2_vscode.zip](https://github.com/ms-iot/vscode-ros/files/9037529/ros2_vscode.zip)
.

closed 2023-03-24 by yeyanlei