ms-iot/vscode-ros #746: [bug]

andrejpadilha2, 2022-06-30:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  Ubuntu 20.0.4
- [ ] ROS 1: Dist
- [x] ROS 2: Foxy

v 0.8.2

# what is the bug
I am following the URDF tutorial of Automatic Addison for ROS 2 (https://automaticaddison.com/how-to-create-a-simulated-mobile-robot-in-ros-2-using-urdf/, the provided URDF renders in RVIZ but won't render on VS Code. It only shows a "black sphere":

[image]

## Repro steps

1. the URDF file that I am using is the one used in the tutorial -> https://drive.google.com/file/d/1UNBKEZWsgo3cp1XBgdU3UY8TGRS-fMK9/view
2. then I just opened VS Code from my ROS workspace
3. clicked on the URDF, and then used "URDF preview"

# expected behavior
I would expect that it rendered a small diff robot, like it does on Rviz.

Thank you so much, I am using VS Code actually because of the great content from @ooeygui at the "A polyhobbyst".

ooeygui, 2022-06-30:
Thank you for the report - and the link to the tutorial. I'll walk through it and see what's up.

ooeygui, 2022-06-30:
I see what the issue is. The Automatic Addison code has a xacro file with a urdf extension. VSCode ROS Extension doesn't run xacro on a urdf; it requires the file have a xacro extension. RosWebTools is ignoring the xacro: xml prefixes so you end up with the bad rendering.

In other words, `basic_mobile_robot.urdf` should be named `basic_mobile_robot.urdf.xacro` or just `basic_mobile_robot.xacro`. 

The reason this works in rviz, is that the automatic addison launch file manually calls xacro on a urdf. The VSCode extension doesn't process launch files in order to render the URDF.

ooeygui, 2022-06-30:
I'm thinking if calling xacro on a urdf is going to generically cause issues; otherwise I might simplify that codepath...

andrejpadilha2, 2022-07-01:
Thank you so much, it works now! The "nav2" official documentation tutorial also uses the .urdf file extension, even using xacro inside it. So I fixed that as well.

[image]

andrejpadilha2, 2022-07-01:
This will help me A LOT to quickly design the code for the robot description!!

ooeygui, 2022-07-01:
Thanks for the feedback! I'm glad you are finding it useful.

I'm going to change this to a feature request to call xacro always; with an option to disable it in case that causes issues.

ooeygui, 2022-10-13:
Related to https://github.com/ms-iot/vscode-ros/issues/97

ooeygui, 2023-05-04:
This has been implemented and pre-released to 0.9.4

closed 2023-05-04 by ooeygui