ms-iot/vscode-ros #753: [bug]

andrejpadilha2, 2022-07-12:
- [ ] Windows: (Version)
- [x] Linux:  20.0.4
- [ ] ROS 1: Dist
- [x] ROS 2: Foxy

Version of the plugin: 0.8.3

I am following the Youtube videos and have managed to install VS Code ROS extension, using locally. I have also been able to SSH into my Jetson Nano, and managed to SSH and enter a docker container of my Jetson Nano, as of the final part of this video: https://www.youtube.com/watch?v=JbBMF1aot5k

But it doesn't source the Foxy setup.bash file. I have to manually source it when I open a new terminal. I have entered the settings of the extension and there are three users (?), and each one of them have "foxy" under "Ros: Distro", but nothing was specified under "Ros Setup Script". I tried to modify these and add "/opt/ros/foxy/setup.bash" but it still won't work. I don't know if it's trying to source an inexistent setup.bash file of the Jetson Nano (outside the container) when it should actually source the setup.bash that lives inside the container.

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. You have a local and remote machine with VS Code and VS Code server installed.
2. You download the ROS Foxy container into the remote computer, from: https://github.com/polyhobbyist/jetbot
3. You can succesfully SSH into remote computer and enter the a folder with a docker container, so that VS Code asks to reopen it using container. You do so.
4. When you open a new terminal (CTRL+SHIFT+P -> New terminal window) and try to run any ros2 command, it returns `bash: ros2: command not found`

# expected behavior
I expected it to source ROS2 Foxy just as it does when I am using my local machine.

polyhobbyist, 2022-07-13:
I'll take a look at it ASAP. Thank you for reporting!

ooeygui, 2022-10-04:
Have you installed the VSCode ROS extension into the container?

andrejpadilha2, 2022-10-05:
> Have you installed the VSCode ROS extension into the container?

I did (at least I think). 

Just to double check: In my VSCode, when I am inside the container, I can see "ROS2.foxy" next to the "Dev Container: jetbot @ ssh://jetson". Also, if I navigate to Extensions>ROS, it says "Extension is enabled on 'Dev Container: jetbot @ssh://jetson"

But If I click on "ROS2.foxy" it shows "ROS2 System Status - offline" in a separate tab.

ooeygui, 2022-10-11:
Mental Note: This seems to be an issue with the login shell used in the container. (Interactive vs not)

ooeygui, 2023-05-04:
This has been corrected and pre-released with 0.9.4

closed 2023-05-04 by ooeygui