ms-iot/vscode-ros #754: [bug] Debugging with ROS extension fails. "Timed out waiting for launcher to connect"

shrinivas96, 2022-07-13:
(Please add appropriate labels)
- [x] Windows: (Windows 10 Version 21H2 OS build 19044.1766)
- [x] Linux:  (WSL2 Ubuntu 20.04)
- [x] ROS 1: noetic
- [ ] ROS 2: Dist

\<Version of the plugin\> v0.8.3

\<Copy the Version information from the `Help | About` menu\>
Version: 1.69.1 (system setup)
Commit: b06ae3b2d2dbfe28bca3134cc6be65935cdfea6a
Date: 2022-07-12T08:21:24.514Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Windows_NT x64 10.0.19044

# what is the bug
\<current behavior\>
When trying to debug a node the debug process starts and the error pops up:
"Timed out waiting for launcher to connect"
[image]

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. [Clone repository](https://github.com/shrinivas96/controlled_husy). Build with ```catkin build smb_highlevel_controller```. 
2. Source workspace with ```source devel/setup.bash```.
3. Open folder in VS Code
4. Use command palette to start ROS with ```Ctrl+Shift+P``` and type ROS: Start.
5. Again, from command palette click on "ROS: Update C++ properties"
6. Go to "Run and Debug" tab and create new ```launch.json``` file. 
7. Select Debugger as "ROS" and select request type as "ROS: Launch" and select package and launch file.
8. Place a breakpoint in the main ros node, and launch debugging with F5.

# expected behavior
\<what you would expect to happen\> To start normal debugging.

# additional context
\<any additional information would be helpful for a better understanding\>
Unable to think of what extra information I could add at the moment. But please let me know when you need any extra information and I will post it.

ooeygui, 2023-05-04:
We were not capturing variables correctly. This has been fixed in 0.9.4-pre

mewescott, 2025-02-11:
I am seeing this again with ROS 2 humble and a lifecycle managed node. 

[image]

ooeygui, 2025-02-11:
@mewescott Thanks for the comment - I'll take a look. I'm in the process of breaking apart the extension into a ROS 1 and ROS 2 version so that I can independently update - which will also improve the ability to downlevel support both. I'll post when this is ready.