ms-iot/vscode-ros #760: `ros.rosSetupScript` seems not to have any affect

senceryazici, 2022-07-24:
- [X] Linux:  (ubuntu/20.04)
- [X] ROS 1: noetic

VSCode version:
```
Version: 1.69.2
Commit: 3b889b090b5ad5793f524b5d1d39fda662b96a2a
Date: 2022-07-18T16:14:10.636Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Linux x64 5.17.5-76051705-generic
```
ROS extension version:
```
v0.8.3
```

# `ros.rosSetupScript` seems not to have any affect
URDF preview fails to find package path.
```
Command failed: bash -c "xacro '/home/user/catkin_ws/src/package1/urdf/robot.xacro' && \ env" 

error: resource not found: package1 
ROS path 
[0]=/opt/ros/noetic/share/ros ROS path
[1]=/opt/ros/noetic/share
when processing file: /home/user/catkin_ws/src/package1/urdf/turquoise.xacro
```
I use zsh, and the workspace is configured with install space. I use 
```
catkin build
```
for build.
Even though I set `rosSetupScript` in `settings.json` as below, 
```json
    ...
    "ros.distro": "noetic",
    "ros.rosSetupScript": "/home/user/catkin_ws/install/setup.bash",
    "ros.isolateEnvironment": "false"
    ... 
```
it's not able to find the ros package. The result is also same if a new terminal inside VSCode is created via `ROS: Create Terminal` command. Is it possible that `ros.rosSetupScript` does not have affect?

`Note:` The workspace is **NOT** sourced by default, when a new terminal opens.
But I'm sourcing the workspace preceding to launching vscode from a terminal.

`Note2:` Also tried to put sourcing script in both `.zshrc` and `.bashrc` to source at terminal startup

`Note3:` Also tried to perform same actions from a bash rather than zsh.

`Note4:` Also tried to perform same actions with zsh in `settings.json`
```json
    ...
    "ros.distro": "noetic",
    "ros.rosSetupScript": "/home/user/catkin_ws/install/setup.zsh",
    "ros.isolateEnvironment": "false"
    ... 
```

## reproduce method 1
1. Open a xacro file as active file
2. configure `settings.json` as described above
3. run `ROS: Preview URDF` command

## reproduce method 2
1. run `ROS: Create Terminal`
2. run `rospack find package1`

# expected behavior
The package should be in the ros package path.

Thanks.

ct2034, 2022-10-18:
I am having the same issue

ooeygui, 2023-05-04:
I was not capturing the variables correctly. This has been fixed in 0.9.4-pre

closed 2023-05-04 by ooeygui