ms-iot/vscode-ros #765: Debugger do not stop at break points

MISSEY, 2022-08-02:
(Please add appropriate labels)
- [x ] Linux: Ubuntu 20.04.4 LTS (Focal Fossa)"  (Dist/Version)
- [x ] ROS 1: noetic

\<Version of the plugin\>

ROS 0.8.3

\<Copy the Version information from the `Help | About` menu\>

Version: 1.69.2
Commit: 3b889b090b5ad5793f524b5d1d39fda662b96a2a
Date: 2022-07-18T16:14:10.636Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Linux x64 5.11.0-46-generic

# what is the bug
\<current behavior\>

I have setup the debugger as described in the [video](https://www.youtube.com/watch?v=uqqHgYsskJI). 

But somehow the debugger is not stopping at breakpoints. 

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>

1.tasks.json

```json
{
    "version": "2.0.0",
    "tasks": [
        {
            "label": "make_debug",
            "type": "catkin_make",
            "args": [
                "--directory",
                "${workspaceFolder}",
                "-DCMAKE_BUILD_TYPE=RelWithDebInfo"
            ],
            "problemMatcher": [
                "$catkin-gcc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
        {
            "label": "make_debug_and_core",
            "dependsOn": [
                "make_debug",
                "roscore: roscore", // This task is provided by vscode-ros
            ]
        },
    ]
}
```
2. launch.json

```json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Launch",
            "request": "launch",
            "target": "/media/sami02/1E4B5BB258FA2EEC/projects/ag_pcl_edge_detection/src/ros_pcl_implementer/launch/main.launch",
            "launch": [
                "rviz",
                "gz",
                "gzclient",
                "gzserver"
            ],
            "console": "integratedTerminal",
            "preLaunchTask": "make_debug_and_core",
            "type": "ros"
        }
    ]
}
```
3. settings.json

```json
{
    "python.autoComplete.extraPaths": [
        "/media/sami02/1E4B5BB258FA2EEC/projects/ag_pcl_edge_detection/devel/lib/python3/dist-packages",
        "/media/sami02/1E4B5BB258FA2EEC/projects/chargepal/mujoco_dope/devel/lib/python3/dist-packages",
        "/opt/ros/noetic/lib/python3/dist-packages"
    ],
    "files.associations": {
        "functional": "cpp"
    },
    "cmake.configureOnOpen": false,
    "ros.distro": "noetic"
}

```

# expected behavior
\<what you would expect to happen\>

I was expecting to stop at line number 9, but it didn't.

# additional context
\<any additional information would be helpful for a better understanding\>

[image]

[image]

ooeygui, 2022-08-02:
Hi there,
Everything looks setup right - however, It looks like your code has not been built with Symbols. I see you are using your own pre-launch task `"preLaunchTask": "make_debug_and_core",` which I suspect is not building `catkin_make --DCMAKE_BUILD_TYPE=RelWithDebInfo` or ` catkin_make --DCMAKE_BUILD_TYPE=Debug`

MISSEY, 2022-08-03:
Hi @ooeygui ,

Thank you for your response,

When nothing worked, I followed this:
https://github.com/ms-iot/vscode-ros/blob/master/doc/debug-support.md#-use-tasks-to-automatically-build-before-starting-debug-session

Should I delete the builds and let build from tasks ?

MISSEY, 2022-08-03:
Now, when I build again, this is output, again it do not stop on breakpoints. 

#### steps 

* ctrl + shift + p
* Tasks: Run Task
* make_debug_and_core

[image]

```json
{
    "version": "2.0.0",
    "tasks": [
        
        {
            "label": "make_debug_and_core",
            "dependsOn": [
                "make_debug",
                "roscore: roscore", // This task is provided by vscode-ros
            ]
        },

        {
            "label": "make_debug",
            "type": "catkin_make",
            "args": [
                "--directory",
                "${workspaceFolder}",
                "-DCMAKE_BUILD_TYPE=RelWithDebInfo"
            ],
            "problemMatcher": [
                "$catkin-gcc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
        
    ]
}
```

ooeygui, 2022-08-03:
oh... I think there's a bug in that file... the DCMAKE_BUILD_TYPE should have 2 dashes before it. Can you try that really quick? If that works, I'll update the doc.

MISSEY, 2022-08-04:
Hi @ooeygui,

It worked, I build using ctrl + shift + P, tasks after deleting the previous build. 

Here are my setting files

### tasks.json

```json
{
    "version": "2.0.0",
    "tasks": [
        
        {
            "label": "make_debug_and_core",
            "dependsOn": [
                "make_debug",
                "roscore: roscore", // This task is provided by vscode-ros
            ]
        },

        {
            "label": "make_debug",
            "type": "catkin_make",
            "args": [
                "--directory",
                "${workspaceFolder}",
                "-DCMAKE_BUILD_TYPE=RelWithDebInfo"
            ],
            "problemMatcher": [
                "$catkin-gcc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
        
    ]
}
```

### settings.josn

```json
{
    "python.autoComplete.extraPaths": [
        "/media/sami02/1E4B5BB258FA2EEC/projects/ag_pcl_edge_detection/devel/lib/python3/dist-packages",
        "/media/sami02/1E4B5BB258FA2EEC/projects/chargepal/mujoco_dope/devel/lib/python3/dist-packages",
        "/opt/ros/noetic/lib/python3/dist-packages"
    ],
    "files.associations": {
        "functional": "cpp"
    },
    "cmake.configureOnOpen": false,
    "ros.distro": "noetic"
}
```

### launch.json

```json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Launch",
[... 241 lines cut ...]
#   FILES_MATCHING PATTERN "*.h"
#   PATTERN ".svn" EXCLUDE
# )

## Mark other files for installation (e.g. launch and bag files, etc.)
# install(FILES
#   # myfile1
#   # myfile2
#   DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}
# )

#############
## Testing ##
#############

## Add gtest based cpp test target and link libraries
# catkin_add_gtest(${PROJECT_NAME}-test test/test_ros_pcl_implementer.cpp)
# if(TARGET ${PROJECT_NAME}-test)
#   target_link_libraries(${PROJECT_NAME}-test ${PROJECT_NAME})
# endif()

## Add folders to be run by python nosetests
# catkin_add_nosetests(test)

```

closed 2022-08-04 by MISSEY