ms-iot/vscode-ros #784: [bug]

bernatgaston, 2022-09-05:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Dist/Version)
- [ ] ROS 1: Dist
- [x] ROS 2: Dist

\<Version of the plugin\>

0.8.3

# what is the bug
I am new using this plugin so maybe there is something I am not getting correctly. However, it's been a while that I am trying to find a response and I am unable to find it.
I can add breakpoints to the code, however, those inside a while are "moved" to the end of the while when the debug starts. Also, if you try to hot-add them (with the debug already running), it does not allow you to do it.

## Repro steps
https://github.com/bernatgaston/hello_world_ros2_node
1. Set up a breakpoint at file src/hello_world.cpp line 12 and another at line 14.
2. Execute debugging from Run and Debug -> ROS:Launch
3. See that line 12 breakpoint is ok, but line 14 breakpoint has been moved to line 18
4. See that you can not add any breakpoint inside the while

# expected behavior
Be able to set breakpoints at any code line

# additional context
Attaching a video

Thank you for all your work!

https://user-images.githubusercontent.com/6225187/188482572-9fa1b30d-4fa8-4a6b-99ba-c30a236c2154.mp4

ooeygui, 2022-10-10:
This is a function of using optimized code. The VSCode ROS extension will build using DebWithRelInfo. Release builds are optimized; so the debugger jumps around.

I changed this to a feature request to support changing the cmake build type - allowing you to pass 'Debug', which disables the optimizer.

bernatgaston, 2022-10-10:
It makes sense... Thanks!