ms-iot/vscode-ros #804: [bug] Launch not working even after sourcing

igricart, 2022-10-11:
- [x] ROS 1: Noetic
- Linux 20.04.1
- Plugin Version: 0.8.3

Version: 1.72.0
Commit: 64bbfbf67ada9953918d72e1df2f4d8e537d340e
Date: 2022-10-04T23:20:47.539Z
Electron: 19.0.17
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 5.13.0-51-generic snap
Sandboxed: No

# what is the bug
I am having the same issue as in another [Issue](https://github.com/ms-iot/vscode-ros/issues/379). My error is related to a package that couldn't be found in the launch file, defined as
`<include file="$(find belt_loader_bringup)/launch/run.launch" pass_all_args="true"/>`

[image]

I tried opening `vscode` from a sourced terminal (bash or zsh) and none of them worked for me.

igricart, 2022-10-13:
I tried starting vscode from my root folder of the catkin workspace, but it still doesn't work. I will go through the video to check if I am doing something wrong

igricart, 2022-10-13:
I tried adding `  "ros.rosSetupScript": "/home/imr/Documents/ws/tmp/2022-08-05/ps_ws/devel_release/setup.bash"` to the settings and now got this error:
[image]

igricart, 2022-10-13:
Now I updated the `launch.json` to:
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "name": "ROS: Launch",
      "type": "ros",
      "request": "launch",
      "target": "${workspaceFolder}/src/ps_tld/tld_bringup/launch/run.launch",
      "arguments": [
        "rviz:=false"
      ],
      "launch": [
        "rviz"
      ]
    }
  ]
}
```

And it complains about an `.sh` file
[image]

ooeygui, 2023-03-27:
It looks like your launch file has a bash script in it. These are not supported in the current launch file.