ms-iot/vscode-ros #805: [feature] Support Yaml launch files

ooeygui, 2022-10-12:
**what is needed**
Humble supports XML and Yaml launch files - https://docs.ros.org/en/foxy/Tutorials/Intermediate/Launch/Creating-Launch-Files.html.

**why is it needed**
This a core ROS feature

rjoomen, 2023-03-28:
Version 0.8.4 required `.py` or `.xml` launch files, but 0.9.0 only accepts `.py` files. This is a serious issue for developers using xml launch files. 

With the older versions, renaming `.launch` files to `.launch.xml` allowed debugging just fine, but that's not possible anymore in 0.9.0, why?

ooeygui, 2023-03-28:
Oh no
Working on it - this was ony supposed to affect ROS2.

ooeygui, 2023-03-28:
XML wasn't really supported - it was treated as a python script. 
I'll back out that change, but it needs specific work to function property.

ooeygui, 2023-03-28:
Sorry about this - releasing 0.9.1 to address.

ooeygui, 2023-05-04:
A fix for this has been released to  pre-release

closed 2023-05-04 by ooeygui