ms-iot/vscode-ros #813: [bug] Debug configuration not able to attach to ROS2 node or launch it

DerProfessor, 2022-10-18:
- [ ] Windows: (Version)
- [X] Linux:  (Ubuntu/18)
- [ ] ROS 1: Dist
- [X] ROS 2: Foxy (self-compiled and is working)

VSCode Extension for ROS v.0.8.3 (same on target and remote machine)

VSCode Info (same on target and remote machine):
Version: 1.72.2
Commit: d045a5eda657f4d7b676dedbfa7aab8207f8a075
Date: 2022-10-12T22:15:17.426Z
Electron: 19.0.17
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux arm64 4.9.253-tegra
Sandboxed: No

# what is the bug
Trying to attach to the running ros2 node (pure python package) or launching it via the VSCode Debugging Configurations does not work. No output is shown and the debugger does not start. I searched possible outputs in VSCode like for example under ROS or Python but nothing is shown. I can launch the node normal via command line. First I thought it was maybe due to trying to run via SSH-Connection but local execution on the target machine does not work.
My guess is maybe that the python venv I use is not recognized but due to missing outputs I'm unable to confirm this. I extended the python-paths for VSCode and resolution with IntelliSense works as normal.

## Repro steps
1. Creating an attach or launch configuration on the debug page.
2. Setting the process id (for attach) or the corresponding launch file under `install/package/.../*.launch.py`
3. Running the corresponding configuration.

# expected behavior
Node launches and Python debugger is active stopping at the selected break points.

ooeygui, 2022-10-18:
Hi there,
Can you tell me more about the virtual environment? 

VSCode ROS extension tries to initialize ROS by itself; it seems that it is unable to find your virtual environment. 

There are two settings you can try in your launch file, which overrides the extensions default processing of environment:

| Json Option | Setting Name | Description |
|---|:---|---|
ros.rosSetupScript | ROS workspace setup script. Overrides ros.distro. | If specified, this will cause the extension to source this script before generating the launch debugging or ROS terminal environment. This overrides the ros.distro, and can be used to specify user scripts or ROS installs in a different location.
ros.isolateEnvironment | Specify if the extension should not capture the environment VSCode is running in to pass to child processes. | Off by default, This setting will prevent the ROS extension from capturing it's hosting environment in case this would conflict with the ROS environment.

DerProfessor, 2022-10-20:
Hi there,
I installed some dependencies within it that needed to build locally (machine-specific TensorRT stuff). It's a Python virtualenv and I extended the `PYTHONPATH` environment variable and built the workspace with it. I done this because the official ros2 doc entry about [Python virtualenvs](https://docs.ros.org/en/foxy/How-To-Guides/Using-Python-Packages.html#installing-via-a-virtual-environment) does not work in my workspace for some reason and there exists no equivalent of the [catkin_virtualenv](https://github.com/locusrobotics/catkin_virtualenv) package for colcon in ros2 yet.

From the two options you gave I used the first `ros.rosSetupScript` in order to use another workspace to build my local workspace. The source order should something like this: foxy setup -> foxy ws1 setup -> local foxy ws2 setup.

My guess is that maybe the ros extension does not recognize the extended `PYTHONPATH`? Do you have an idea where I could look to see if the venv's packages are included?

Also later I could try to move the venv's packages to `/usr/local` for testing.

DerProfessor, 2022-10-24:
Just wanted to give a quick update. I solved the problems regarding the ROS extension as well as those with the Python environment.
After some digging I realized that the `debugpy` dependency of the main Python VsCode extension [stopped supporting Python 3.6](https://github.com/microsoft/debugpy/releases/tag/v1.6.0), the version I use. There are plans to [separate debugpy into a separate extension](https://github.com/microsoft/vscode-python/issues/19559) but there is no final release so far regarding this issue. Downgrading to version [`2022.2.1924087327`](https://github.com/microsoft/vscode-python/releases/tag/2022.2.1924087327) (the latest to support Python 3.6) resolves this temporarily. Afterwards the debug functionality of the ROS extension finally worked for me. 😊
Regarding the python virtualenv I realized that I was missing the necessary entries for the dependencies in the `package.xml` of my project. Adding these and following the official ros2 guide for [Python venvs](https://docs.ros.org/en/foxy/How-To-Guides/Using-Python-Packages.html#installing-via-a-virtual-environment) again I managed to resolve the dependencies during the execution of the nodes. Extending `PYTHONPATH` is now no longer necessary for me.
Now everything works regarding debugging Python code with the ROS extension. 🎉

closed 2022-10-24 by DerProfessor