ms-iot/vscode-ros #818: [bug] C/C++ standards overridden and invalid for newer ROS2

mjbogusz, 2022-10-25:
- [x] Linux:  Ubuntu 20.04/22.04
- [x] ROS 2: Galactic/Humble/Rolling

\<Version of the plugin\>
`v0.8.3`

# what is the bug
The plugin overrides the C/C++ standards set in `c_cpp_properties.json` with static `gnu11`/`c++14` which are not only arbitrary (GNU extensions for C but not for C++?) but also invalid for ROS2 Galactic/Humble/Rolling, requiring C++17 as per [REP2000](https://www.ros.org/reps/rep-2000.html#galactic-geochelone-may-2021-november-2022)

## Repro steps
1. Set up a workspace with ROS2 Galactic or newer, set C/C++ standards (e.g. `gnu17`/`gnu++17`)
2. Use `ROS: update C++ Properties`

# expected behavior
C/C++ standards should not be changed if set and set to appropriate version only if not previously present.

# additional context
Discussion question: what if the standards are set per-user or per-workspace (this plugin sets them per-folder)?
Discussion question 2: what if the standards are set but to a lower version than required?

BrunoB81HK, 2024-04-10:
Is this going to be implemented?

ooeygui, 2024-04-10:
@BrunoB81HK 
Would love to but have no timeline. Happy to accept a PR for it.

BrunoB81HK, 2024-04-10:
I'll see if I find some time.

ooeygui, 2024-04-10:
@BrunoB81HK Thank you!

PR #1287 (2024-04-27): Fix the `c_cpp_properties.json` configuration overwrite issue (#818)
This change fix #818. It changes only the configurations that are not already set.

One thing that was not changed is the default C++ standard being at C++14.
closed 2024-04-27 by ooeygui