ms-iot/vscode-ros #830: [bug] Cannot use "ROS: Launch" debuger in a docker container for C++ packages (Unable to determine path to debugger)

songanz, 2022-11-22:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  local: 20.04, docker container: 18.04
- [x] ROS 1: in docker container: melodic
- [ ] ROS 2: Dist

\<Version of the plugin\>
v0.8.4, installed and enabled in the container

\<Copy the Version information from the `Help | About` menu\>
VScode
Version: 1.73.1
Commit: 6261075646f055b99068d3688932416f2346dd3b
Date: 2022-11-09T03:54:53.913Z
Electron: 19.0.17
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 5.4.0-131-generic
Sandboxed: No

# what is the bug
After attaching vscode to the running container, I installed the ROS extension. Then I setup the debugger using "ROS: Launch " template. 
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Launch",
            "request": "launch",
            "target": "path/to/.launch",
            "launch": [
                "rviz",
                "gz",
                "gzclient",
                "gzserver"
            ],
            "type": "ros",
        }
    ]
}
```
However, it cannot launch the debugger, with error: Unable to determine path to debugger. Please specify the "MIDebuggerPath" option.
[image]

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Build and run a ROS melodic docker container (for example `docker pull ros:melodic`)
2. Attach to this running container
3. Install the ROS extension and make sure ROS core is running
4. Setup the launch.json for debugging
5. Start debugging

# expected behavior
Debugger successfully started in the container

# additional context
\<any additional information would be helpful for a better understanding\>
I have also setup a docker for ROS2 and have some python packages running in that docker container, and I can start debugging successfully with exactly the same steps.

ooeygui, 2022-11-22:
The ros melodic container does not have gdb installed, so VSCode cannot debug C++ nodes. You can derive a container using a Dockerfile which injects it, or manually install it from a terminal connected to the container. I have a video on how to use the VSCode extension with containers - https://youtu.be/JbBMF1aot5k.

closed 2022-11-22 by ooeygui

ooeygui, 2022-11-22:
I saw your note on my container video. I may have missed the specific issue about injecting gdb - I covered that in the C++ video - https://youtu.be/uqqHgYsskJI

songanz, 2022-11-22:
Thank you so much for your prompt response! I confirm this problem is solved! (dammit, I had a feeling this is trivial)

ooeygui, 2022-11-23:
No worries - I think I have a feature in the list to detect that gdb isn't installed and suggest solutions. The backlog grows faster than I can burn it down!