ms-iot/vscode-ros #844: [Question] Intro guide for existing projects

bearrito, 2022-12-22:
I am to trying to understand how to integrate this plugin with an existing ROS1 project.  If there is already a guide, please point me to that.

I am trying to to start off with a simple package, My setup is as follows (very vanilla ros package):
```
ws/
  custom_package/ 
   / launch
      my.launch
   / src
      /custom_package
        /foo.py -> contain class DoAThing
   / scripts
      myscript.py
  custom_package_msgs/
   /msg
     MyMsg.msg
```
This is all on top of another workspace, with many prebuilt packages , some 3rd party others internally developed.
My CMakelists.txt for both projects are correct. I can catkin build and roslaunch fine, outside of the plugin e.g the below works fine
```
source ..ws1/../devel.bash
source ..ws2/../devel.bash
catkin build
roslaunch custom_package my.launch
```

I am trying to figure out the following two issues.

1.  The modules under src are not available for autocomplete/highlighting in the scripts folder. Again, these import fine when launching or running a node. I'm simply unable to get any form of intellisense when I am working on a script in the scripts folder. In scripts/myscript.py I import as

```
from custom_package.foo import DoAThing
```
This just shows up as non-highlighted.

2. I'm unable to get any intellisense or even type information from msgs in my msgs package

```
from custom_package_msgs.msg import MyMsg
```
Again,  VSCode doesn't recognize that Msg, and has no intellisense highlighting or features

This seems like a basic path issue, but I'm not sure what the plugin is expecting for existing codebases

bearrito, 2022-12-23:
It looks like if I set workspace paths in settings, I can get things working

```
"python.autoComplete.extraPaths": [
    "/home/ws/devel/lib/python3/dist-packages",
    "/home/ws/src/custom_package/src"
  ],
  "python.analysis.extraPaths": [
    "/home/ws/devel/lib/python3/dist-packages",
    "/home/ws/src/custom_package/src"
  ]
```

I can't help but feel this isn't correct, but it does work. Another issue, is that unittests don't seem to discover modules inside src/

So in inside /test/test_custom_package.py I import as follows

```
from custom_package.foo import DoAThing
```

Because of the extra settings above, intellisense can discover those modules. However, vscode test discovery fails.
```
ModuleNotFoundError: No module named 'custom_package'
```

I will point when running tests directly I don't have an issue

```
rosunit test/test_custom_package.py  
``
runs fine

ooeygui, 2023-03-27:
Thanks for your interest in the extension. I made a video on how to use the extension with ROS1 (and many other cases) - https://youtu.be/N2vqBvPQdhE

closed 2023-03-27 by ooeygui