ms-iot/vscode-ros #848: [bug] ROS2 urdf preview does not show stl files

kinami-takashi, 2023-01-08:
- [x] Linux:  ubuntu20.04 
- [x] ROS 2: foxy

ROS plugin version 0.8.4, VScode version 1.74.2

# what is the bug
ROS2 urdf preview does not show stl files

## Repro steps
I am unable to preview urdf files even if I specify the following in the xacro file (I have confirmed that it works fine when I start rviz etc.).
```
<mesh filename="file://$(find PACKAGE_NAME)/urdf/body.stl"  />
```
We know from the following issue that the "file://" path does not work in the ROS plugin, only the "package://" path.
[https://github.com/ms-iot/vscode-ros/issues/183](https://github.com/ms-iot/vscode-ros/issues/183)

However, we know from ROS Answer below that only the "file://" path is supported in ROS2 (we actually tried it, and it did not work with the "package" path).
[https://answers.ros.org/question/381116/cant-load-stl-into-urdf-using-ros2/](https://answers.ros.org/question/381116/cant-load-stl-into-urdf-using-ros2/)

Is there a way to preview urdf in "ROS2" at this time?

atticusrussell, 2023-03-01:
I'm encountering the same issue. The built-in preview is an incredibly convenient feature, but the package:// breaks gazebo.

ooeygui, 2023-03-27:
Could you check with the 0.9.0 release?

ooeygui, 2023-04-17:
I'm rewriting the URDF previewer given the issues with RosWebTools. STL and glTF are implemented, collada is being worked on. Once that is completed and verified, I'll push into 0.9.4

kinami-takashi, 2023-04-17:
OK, I will close this issue once 0.9.4 is released and the problem is resolved.

atticusrussell, 2023-04-21:
Okay so that isn't working for me. Should I file a new bug as suggested in:
https://github.com/ms-iot/vscode-ros/issues/908#issuecomment-1517898578?
or would you like me to describe it here.
I have one robot that is rendering incorrectly, and another that is having a rendering error and not rendering.

ooeygui, 2023-05-04:
A fix for this has been released to pre-release in the marketplace. You can switch to that version using the button on the marketplace page:
[image]

closed 2023-05-04 by ooeygui

belem2050, 2024-06-08:
> A fix for this has been released to pre-release in the marketplace. You can switch to that version using the button on the marketplace page: [image]

Tested and working for me.