ms-iot/vscode-ros #851: Attaching to a running ros2 node does nothing [bug]

fawadshahid, 2023-01-16:
- [ ] Windows: (Version)
- [x] Linux:  (20.04/Version)
- [ ] ROS 1: Dist
- [x] ROS 2: Galactic

\<Version of the plugin\>
0.8.4
\<Copy the Version information from the `Help | About` menu\>

# what is the bug
Attaching does nothing and takes me to the launch.json file.

## Repro steps
1. Add debug configuration. 
2. Add breakpoints.
3. Run node using ros2 launch command (Not debugging the launch file but a src python script).
4. Run debug config

# expected behavior
Stop at the break point.
# additional context
I have also used --cmake-args while building.

ooeygui, 2023-03-27:
Can you point at a repository which reproduces this behavior?

jweaston, 2024-01-17:
I am also having this issue. I am using ROS2 humble. In a Ubuntu 22.04 container running in WSL. I am able to get the ROS: Launch debug functionality but not the attach functionality.

jweaston, 2024-01-18:
> I am also having this issue. I am using ROS2 humble. In a Ubuntu 22.04 container running in WSL. I am able to get the ROS: Launch debug functionality but not the attach functionality.

After some looking around I found this error being output when attempting to use the ROS: Attach debugger function in the Extension Host (Remote) output in VSCode
`[error] Error: Command [python /home/ros/.vscode-server/extensions/ms-python.python-2023.22.1/pythonFiles/lib/python/debugpy --listen localhost:8000 --pid 737] failed!
	at t.AttachResolver.<anonymous> (/home/ros/.vscode-server/extensions/ms-iot.vscode-ros-0.9.6/dist/extension.js:2:499172)
	at Generator.throw (<anonymous>)
	at a (/home/ros/.vscode-server/extensions/ms-iot.vscode-ros-0.9.6/dist/extension.js:2:497653)
	at process.processTicksAndRejections (node:internal/process/task_queues:95:5)`

jweaston, 2024-01-29:
Looks like this might be a duplicate of [this issue](https://github.com/ms-iot/vscode-ros/issues/872)