ms-iot/vscode-ros #869: [bug] Error: Failed to start debug session in Ubuntu 18.04

ravijo, 2023-02-03:
# Version Info
- [ ] Windows: (Version)
- [x] Linux: Ubuntu OS 18.04.6 LTS
- [x] ROS 1: Melodic
- [ ] ROS 2: Dist

\<Version of the plugin\>
Visual Studio Code v1.74.3
VSCode-ROS Extension v0.8.4

\<Copy the Version information from the `Help | About` menu\>
```
Version: 1.74.3
Commit: 97dec172d3256f8ca4bfb2143f3f76b503ca0534
Date: 2023-01-09T16:57:40.428Z
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 5.4.0-137-generic
Sandboxed: No
```

# Bug Info
Can not attach or launch the debugger at all.

## Repro steps
1. Clone [rospy_tutorials](https://github.com/ros/ros_tutorials/tree/melodic-devel/rospy_tutorials)
2. Use the following `launch.json` file:
    ```
    {
        // Use IntelliSense to learn about possible attributes.
        // Hover to view descriptions of existing attributes.
        // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
        "version": "0.2.0",
        "configurations": [
            {
                "name": "ROS: Launch",
                "type": "ros",
                "request": "launch",
                "target": "/home/ravi/test_ws/src/hello_tutorials/talker_listener.launch"
            },
    
            {
                "name": "ROS: Attach",
                "type": "ros",
                "request": "attach"
            }
        ]
    }
    ```

# Expected Behavior
Debugger should run.

# Additional Info
Please see two screen recordings below:

1. Recording: Attaching a running node

https://user-images.githubusercontent.com/1439011/216499976-aff43c6b-8b9b-48cd-8a40-09f6b0df3dc1.mp4

2. Recording: Launching nodes

https://user-images.githubusercontent.com/1439011/216499996-906ac690-1f7c-4da5-883c-7177df0ea3ae.mp4

**I noticed that attaching a node is requesting authentication. But after providing the correct password, nothing happens. Similarly, the launching nodes show processes for a second but die/disappear immediately.**

The developer tools are showing following following error:
```
[Extension Host] rejected promise not handled within 1 second: Error: Failed to start debug session!
[Extension Host] stack trace: Error: Failed to start debug session!
    at LaunchResolver.<anonymous> (/home/ravi/.vscode/extensions/ms-iot.vscode-ros-0.8.4/out/src/debugger/configuration/resolvers/ros1/launch.js:310:32)
    at Generator.next (<anonymous>)
    at fulfilled (/home/ravi/.vscode/extensions/ms-iot.vscode-ros-0.8.4/out/src/debugger/configuration/resolvers/ros1/launch.js:7:58)
    at process.processTicksAndRejections (node:internal/process/task_queues:96:5)
```

Please see a screenshot below:
[image]

The issues is also reported in [SO](https://stackoverflow.com/q/75321557)

ravijo, 2023-02-03:
It seems that error is caused due to Python 2.7 as indicated by the following screenshot:

[image]

Below is the `launch.json` file:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Launch",
            "request": "launch",
            "target": "/home/ravi/test_ws/src/hello_tutorials/talker_listener.launch",
            "launch": [
                "rviz",
                "gz",
                "gzclient",
                "gzserver"
            ],
            "type": "ros"
        }
    ]
}
```

**How to launch debugger in ROS Melodic?**

DerProfessor, 2023-02-05:
Hi,
I encountered the problem already with Python 3.6 under Ubuntu 18.04 (see #813).
In short the problem is that the VSCode Python Extension uses debugpy which dropped support for Python 2.7 and < Python 3.7. I work around the problem by reinstalling version [2022.2.1924087327
](https://github.com/microsoft/vscode-python/releases/tag/2022.2.1924087327) of the Python Extension which is the last to support Python 2.7 and 3.6. There are [plans to support older Python versions](https://github.com/microsoft/vscode-python/issues/19559) again.

ravijo, 2023-02-06:
@DerProfessor 

It works like a charm! Thank you very much.

I hope the plans to support older versions get a green signal. Until then, I will keep using [2022.2.1924087327](https://github.com/microsoft/vscode-python/releases/tag/2022.2.1924087327), as you suggested.

Thanks again.

ravijo, 2023-02-10:
@DerProfessor 

On another computer, having same OS and configuration, the "Launch" works but "Attach" doesn't while debugging python node.

May I request you to please have a look at #872

BTW, is there any log file in vscode? It would help to dig deeper in these issues. Thanks

ooeygui, 2023-03-27:
There is a new log file support in 0.9.0. Using the "output" tab, select "ROS", which now shows logging. (Before it was only showing when debugged).

That said, these issues are related to the Python extension, not the ROS extension. Because of this dependency, Melodic is not supported by current versions of the ROS extension.

closed 2023-03-27 by ooeygui

ravijo, 2023-03-28:
@ooeygui 

> There is a new log file support in 0.9.0. Using the "output" tab, select "ROS", which now shows logging. (Before it was only showing when debugged).

Thank you very much for this helpful information.

> That said, these issues are related to the Python extension, not the ROS extension. Because of this dependency, Melodic is not supported by current versions of the ROS extension

I understand. If you don't mind, may I ask you for a workaround please?