ms-iot/vscode-ros #872: [bug] "Launch" works but "Attach" doesn't while debugging python node

ravijo, 2023-02-10:
# Version Info
- [ ] Windows: (Version)
- [x] Linux: Ubuntu OS 18.04.6 LTS
- [x] ROS 1: Melodic
- [ ] ROS 2: Dist

\<Version of the plugin\>
Visual Studio Code v1.75.0
VSCode-ROS Extension v0.8.4

\<Copy the Version information from the `Help | About` menu\>
```
Version: 1.75.0
Commit: e2816fe719a4026ffa1ee0189dc89bdfdbafb164
Date: 2023-02-01T15:29:17.766Z
Electron: 19.1.9
Chromium: 102.0.5005.194
Node.js: 16.14.2
V8: 10.2.154.23-electron.0
OS: Linux x64 5.4.154-rt65
Sandboxed: No
```

# Bug Info
"Launch" works but "Attach" doesn't while debugging python node

## Repro steps
1. Clone [rospy_tutorials](https://github.com/ros/ros_tutorials/tree/melodic-devel/rospy_tutorials)
2. Use the following `launch.json` file:
    ```
    {
        // Use IntelliSense to learn about possible attributes.
        // Hover to view descriptions of existing attributes.
        // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
        "version": "0.2.0",
        "configurations": [
            {
                "name": "ROS: Attach",
                "type": "ros",
                "request": "attach"
            }
        ]
    }
    ```
3. Run `roscore` and python node via `rosrun`
4. Attach the python node to debugger

# Expected Behavior
Debugger should run.

# Additional Info
Please see following screen recording/ It shows that "Attach" is not working, though "Launch" is working fine.

https://user-images.githubusercontent.com/1439011/217989851-f3a7bcd2-167d-4d40-b759-609d269c60dd.mp4

The developer tools are showing following following error:
```
Error: Command [python /home/ravi/.vscode/extensions/ms-python.python-2022.2.1924087327/pythonFiles/lib/python/debugpy --listen localhost:8000 --pid 2609] failed!
	at AttachResolver.<anonymous> (vscode-file://vscode-app/home/ravi/.vscode/extensions/ms-iot.vscode-ros-0.8.4/out/src/debugger/configuration/resolvers/attach.js:97:28)
	at Generator.throw (<anonymous>)
	at rejected (vscode-file://vscode-app/home/ravi/.vscode/extensions/ms-iot.vscode-ros-0.8.4/out/src/debugger/configuration/resolvers/attach.js:8:65)
```

Please see a screenshot below:

[image]

ravijo, 2023-02-20:
Any suggestions, please?

BTW is there any log file in vscode to dig deeper into this issue?

Thanks

xiaomi2077, 2023-03-04:
I have the same problem...

ooeygui, 2023-03-27:
The current version of the VSCode Python extension - which the ROS node depends on - does not support Python 2.7, so I'm surprised that launch works. 

Can you confirm on noetic?

ravijo, 2023-03-28:
@ooeygui 

> The current version of the VSCode Python extension - which the ROS node depends on - does not support Python 2.7, so I'm surprised that launch works.

Yes, you are right. With the default setup (VSCode Python Extension with debugpy), even the launch does not work. However, based on [this suggestion,](https://github.com/ms-iot/vscode-ros/issues/869#issuecomment-1418250077) I reinstalled [VSCode Python Extension 2022.2.1924087327](https://github.com/microsoft/vscode-python/releases/tag/2022.2.1924087327). **This is why launch is working with Python 2.7**

The problem is that only "launch" is wotking, but not "attach" though. As a workaround, I am executing nodes via "launch", but "attach" is a better option though!

Thanks

martinakos, 2023-04-21:
I have same issue with version v0.9.2 and ROS humble. I can launch and stops in a breakpoint but I can't attach.  Any progress on this?

ooeygui, 2023-04-21:
I have not made progress on this. I've added the "help wanted" flag, if someone wants to take a look at this before I get to it, I'm happy to guide. I'll look into python debugging once I stabilize and release URDF rendering

jweaston, 2024-01-29:
After some research I found you can do a similar thing with just the base python debugger with this launch config

```
    {
      "name": "Python Debugger: Current File with Arguments",
      "type": "debugpy",
      "request": "launch",
      "program": "${file}",
      "console": "integratedTerminal",
      "args": "${command:pickArgs}"
    },
```
So far this has worked as I expected the ROS Attach debugger to work.